KCL-Planning/ROSPlan #1: Minor compilation and execution issues

NoWiS-, 2015-05-08:
Hi,

I had some minor problems for compiling that I though I should share:

The first one is the cola2 interface, I had to either clone the cola2 repo or simply make catkin ignore this interface

    $ touch ROSPlan/rosplan_interface_cola2/CATKIN_IGNORE

And some minor dependencies issue to compile with several job, I simply added it in ROSPlan/rosplan_planning_system/CMakeLists.txt

    add_dependencies(planner ${catkin_EXPORTED_TARGETS})

and to ROSPlan/rosplan_knowledge_base/CMakeLists.tx

    add_dependencies(knowledgeBase ${catkin_EXPORTED_TARGETS})

in order to force catkin to generate the knowledge_msgs before building these two modules.

And finally, in order to make the example script work (turtlebot_explore.bash), I had to increase the number of total_attempts (otherwise no waypoints were created) and change "turtlebot_playground.launch" to "turtlebot_world.launch" in rosplan_demos/launch/turtlebot.launch (that's probably a difference between hydro and indigo).  

Thanks for sharing this tool,
Simon Vernhes.

m312z, 2015-05-08:
"The first one is the cola2 interface, I had to either clone the cola2 repo or simply make catkin ignore this interface"

Yes. We will move the interfaces into separate packages and repositories this week.

"And some minor dependencies issue to compile with several job, I simply added it in ROSPlan/rosplan_planning_system/CMakeLists.txt"

YES! I've been looking for this solution for a while. Thank you.

"And finally,..."

I've fixed the script. The launch file name is indeed a hydro/indigo difference.

I will close this issue once the interface packages are sorted out.

m312z, 2015-05-08:
I've moved the cola2 interface components to their own repository.

@NoWiS- Thanks for the suggestions. We're still actively building ROSPlan, so if there are any more bugs or missing features, please let us know.

closed 2015-05-08 by m312z