KCL-Planning/ROSPlan #4: Compilation Error in occupancy_grid_utils

karpase, 2015-06-15:
When compiling, I got this error from occupancy_grid_utils:

/usr/bin/ld: warning: libtiff.so.4, needed by /usr/local/lib/libopencv_highgui.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libavcodec.so.53, needed by /usr/local/lib/libopencv_highgui.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libavformat.so.53, needed by /usr/local/lib/libopencv_highgui.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libavutil.so.51, needed by /usr/local/lib/libopencv_highgui.so, not found (try using -rpath or -rpath-link)

I am running Ubuntu 14.10 with ROS Indigo.
In order to fix this, I had to compile opencv from scratch, passing   -DBUILD_TIFF=ON   to cmake.

m312z, 2015-06-15:
Thanks for the report!

We're currently using occupancy_grid_utils for ray_tracing and coordinate_conversion in the rosplan_interface_mapping package.

Do you mind if I post this issue across to their repository here (or you can)?
https://github.com/clearpathrobotics/occupancy_grid_utils

karpase, 2015-06-15:
Hi Michael,
Feel free to post this anywhere. I am not really an expert with all the
fancy features of github, so I'll leave it to you.

Cheers,
Erez.

closed 2015-10-06 by m312z