KCL-Planning/ROSPlan #5: Unable to Run Turtlebot Demo

karpase, 2015-06-15:
When running the turtlebot demo (following the instructions in the wiki), I get errors in planning. Tracing this, I discovered that the problem PDDL contains objects for waypoints 0..5 (although I noticed this varies - sometimes there one waypoint more or less), but the goal contains waypoints 0..9.

My best guess is that some waypoints are sampled, found to be in collision with an object, and dropped, but still remain in the goal.

The full problem.PDDL follows:

(define (problem turtlebot_demo_task)
(:domain turtlebot_demo)
(:objects
    kenny - robot
    wp0 wp1 wp2 wp3 wp4 wp5 - waypoint
)
(:init
    (connected wp0 wp1)
    (connected wp0 wp2)
    (connected wp0 wp3)
    (connected wp0 wp4)
    (connected wp0 wp5)
    (connected wp1 wp0)
    (connected wp1 wp2)
    (connected wp1 wp4)
    (connected wp1 wp5)
    (connected wp2 wp0)
    (connected wp2 wp1)
    (connected wp2 wp4)
    (connected wp2 wp5)
    (connected wp3 wp0)
    (connected wp3 wp5)
    (connected wp4 wp0)
    (connected wp4 wp1)
    (connected wp4 wp2)
    (connected wp5 wp0)
    (connected wp5 wp1)
    (connected wp5 wp2)
    (connected wp5 wp3)
    (robot_at kenny wp0)
    (= (distance wp0 wp1) 2)
    (= (distance wp0 wp2) 1.45382)
    (= (distance wp0 wp3) 3.72324)
    (= (distance wp0 wp4) 2)
    (= (distance wp0 wp5) 0.997278)
    (= (distance wp1 wp0) 2)
    (= (distance wp1 wp2) 2)
    (= (distance wp1 wp4) 1.69919)
    (= (distance wp1 wp5) 2.99611)
    (= (distance wp2 wp0) 1.45382)
    (= (distance wp2 wp1) 2)
    (= (distance wp2 wp4) 2.98825)
    (= (distance wp2 wp5) 2)
    (= (distance wp3 wp0) 3.72324)
    (= (distance wp3 wp5) 4.68982)
    (= (distance wp4 wp0) 2)
    (= (distance wp4 wp1) 1.69919)
    (= (distance wp4 wp2) 2.98825)
    (= (distance wp5 wp0) 0.997278)
    (= (distance wp5 wp1) 2.99611)
    (= (distance wp5 wp2) 2)
    (= (distance wp5 wp3) 4.68982)
)
(:goal (and
    (visited wp0)
    (visited wp1)
    (visited wp2)
    (visited wp3)
    (visited wp4)
    (visited wp5)
    (visited wp6)
    (visited wp7)
    (visited wp8)
    (visited wp9)
)))

m312z, 2015-06-15:
Your guess is exactly correct.

The demo script file requests 10 waypoints from the roadmap generator and then adds goals. (rosplan_demos/scripts/turtlebot_explore.bash)

The roadmap generator does not always manage to find the required number of waypoints. (We are still working on it.)

Fiddling with the parameters fixes this -- in particular, increasing "total_attempts" will increase the likelihood of finding 10 waypoints.

For now I'll change the demo script with better parameter values.

I'll close this issue once we've revamped rosplan_interface_mapping. Also, I will alter the demo script to add goals that match the actual waypoints generated.

karpase, 2015-06-15:
Thanks, increasing the number of attempts did work for me.

Cheers,
Erez.

closed 2015-12-17 by m312z