KCL-Planning/ROSPlan #10: Install guide did not work.

Napoleon-BlownApart, 2015-10-06:
[Ubuntu 14.04.1 fully updated, ROS Indigo]

I followed the instructions but when I built using 'catkin_make' I got an error informing me that I didn't have GLUT.  So I installed freeglut3 (which apt-get told me I already had) and freeglut3-dev.

I reran catkin_make but got another error, now to do with 'move_base_msgs':

    CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
      Could not find a package configuration file provided by "move_base_msgs"
      with any of the following names:
    
        move_base_msgsConfig.cmake
        move_base_msgs-config.cmake
    
      Add the installation prefix of "move_base_msgs" to CMAKE_PREFIX_PATH or set
      "move_base_msgs_DIR" to a directory containing one of the above files.  If
      "move_base_msgs" provides a separate development package or SDK, be sure it
      has been installed.
    Call Stack (most recent call first):
      rosplan/rosplan_interface_movebase/CMakeLists.txt:5 (find_package)
    
    
    -- Configuring incomplete, errors occurred!

`rospack` says it can't find the package:

    rospack find move_base_msgs
    [rospack] Error: package 'move_base_msgs' not found

Any suggestions?

Cheers,
Nap

m312z, 2015-10-06:
Hello,

Thanks for the heads up, I will add freeglut3-dev to the instructions.

Can you check if you have the move_base_msgs package installed? (http://wiki.ros.org/move_base_msgs)
It is included in an install of ros-indigo-desktop-full, but maybe not the smaller installations. If not, you can get it through "ros-indigo-navigation"

Napoleon-BlownApart, 2015-10-06:
@m312z, thanks for the incredibly quick response.  I `sudo apt-get install ros-indigo-navigation` and it said `move_base_msgs` was a new package.  So it looks like I didn't get it while installing Desktop Full.
I installed it in Feb, when there were some issues with installing Indigo on 14.04.  
http://answers.ros.org/question/203610/ubuntu-14042-unmet-dependencies-similar-for-14043/

lol, now I have a problem with kobuki_msgs.  [Never been a fan of Kobuki theatre... ;)]

m312z, 2015-10-06:
ROSPlan does itself does not depend on Kobuki_msgs, only the rosplan_interface_turtlebot2 package. If you're not using the turtlebot (with a Kobuki base) then you should be able to get by without.

If you do need the messages, try following this installation page:
http://wiki.ros.org/turtlebot/Tutorials/indigo/Debs%20Installation

There are a few packages there which aren't included in desktop-full.

Napoleon-BlownApart, 2015-10-06:
I installed the Turtlebot and now it's all ok.
Thanks for your help.

Napoleon-BlownApart, 2015-10-06:
I already have Turtlebot Gazebo and Turtlebot Simulator installed.  But when I `roslaunch rosplan_demos turtlebot.launch`, I get the following error:

    while processing /opt/ros/indigo/share/turtlebot_gazebo/launch/turtlebot_playground.launch:
    Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/opt/ros/indigo/share /turtlebot_gazebo/launch/turtlebot_playground.launch'
    The traceback for the exception was written to the log file

I will see if I can figure out what's wrong.

m312z, 2015-10-06:
Ah, I think that's a hydro/indigo change. turtlebot_playground is now turtlebot_world.

I will create a hydro branch, and update master to indigo filenames. (For now you can simply change "playground" to "world" in turtlebot.launch -- I think that is the only change to make.

Napoleon-BlownApart, 2015-10-06:
From the ROSPlan main readme page, the section on Running a demo with the turtlebot produces a completely different analysis.

Where can I get the `turtlebot_playground.launch` file?  Or what is its contents?

Napoleon-BlownApart, 2015-10-06:
The above analysis has been running for over an hour now, and it doesn't look anything like what's depicted on the readme page where the guide is.  I guess at some point this analysis will finish and then I'll need to execute the second command `sh src/rosplan_demos/scripts/turtlebot_explore.bash`.

I'm running this on a 2014 MacBook Pro 15" which has an i7 and 16GB mem dual booted with Ubuntu.  How long do  you think this will run for?

Cheers,
Nap

m312z, 2015-10-06:
Hello,

That screenshot looks right. ROSPlan will continue running in the window until you kill it. Once everything is running, you will need to open a new window (or put the other processes into the background) in order to run 'sh src/rosplan_demos/scripts/turtlebot_explore.bash'

Remember to source the setup.bash before running the script.

The script file simply generates some waypoints, adds visiting them as goals, and asks the planning system to start. I have it running here, but there are some things to note:

1) Before running roslaunch rosplan_demos turtlebot.launch, make sure there are no existing mongodb services already running on the machine ('sudo service mongodb stop')

2) The waypoint generation is a bit messy right now, and usually fewer than 9 waypoints are generated. It is best to adjust the goal so that only four are asked to be visited. You can do this by editing the script and altering the second loop. If you get errors from the PDDL problem/domain, this is probably the issue. It only matters in the demo.

3) Gazebo only works 50% of the time. I have to run gazebo in a separate terminal, and reboot it often. (From your screenshot it looks like you don't have this problem.)

4) You might find it useful to use the graphical user interface -- take a look at the wiki here on github. There is a page that shows how to start it. The GUI will show you the current goals and objects -- so if there is a problem with waypoints as in (2) then you can fix it directly.

Michael

Napoleon-BlownApart, 2015-10-06:
Thanks for your help Michael.  I will close the issue then.
Cheers,
Nap.

Napoleon-BlownApart, 2015-10-06:
Hi Michael,
  I decided to wait for the turtlebot to perform a 360 degree scan.
  The path to `src/rosplan_demos/scripts/turtlebot_explore.bash`  is incorrect.  The leading 'src/' is not required, and the script, with the correct path needs to be run from the 'rosplan' folder.  (Also, the script needs chmod +x applied to it.)

Fixing the above, I got the following error when I ran it:

    ERROR: Service [/kcl_rosplan/roadmap_server/create_prm] is not available.

I looked back into the terminal output of the `roslaunch rosplan_demos turtlebot.launch` command and noticed that the following error (concerning mongodb) was showing):

    [ERROR] [WallTime: 1444145988.169251] [8.510000] Can't connect to MongoDB server. Make sure mongodb_store/mongodb_server.py node is started.
    Traceback (most recent call last):
      File "/opt/ros/indigo/lib/mongodb_store/message_store_node.py", line 236, in <module>
        store = MessageStore()
      File "/opt/ros/indigo/lib/mongodb_store/message_store_node.py", line 35, in __init__
        raise Exception("No Datacentre?")
    Exception: No Datacentre?
    [rosplan_scene_message_store-18] process has died [pid 6668, exit code 1, cmd /opt/ros/indigo/lib/mongodb_store/message_store_node.py __name:=rosplan_scene_message_store __log:=/home/user/.ros/log/15a8efc4-6c21-11e5-855d-60f81da9c0e4/rosplan_scene_message_store-18.log].
    log file: /home/user/.ros/log/15a8efc4-6c21-11e5-855d-60f81da9c0e4/rosplan_scene_message_store-18*.log

I already stopped mongodb as you pointed out before running the command.  Trying the launch command another time produced a different error:

    [ERROR] [WallTime: 1444146461.441162] [0.000000] Can't find database at supplied path /home/user/Documents/thesis/ROSPlan/catkin_ws/src/rosplan/rosplan_knowledge_base/common/mongoDB. If this is a new DB, create it as an empty directory.
    [rosplan_scene_database-17] process has died [pid 12581, exit code 1, cmd /opt/ros/indigo/lib/mongodb_store/mongodb_server.py __name:=rosplan_scene_database __log:=/home/user/.ros/log/15a8efc4-6c21-11e5-855d-60f81da9c0e4/rosplan_scene_database-17.log].
    log file: /home/user/.ros/log/15a8efc4-6c21-11e5-855d-60f81da9c0e4/rosplan_scene_database-17*.log

I checked the knowledge_base folder but there was no sub-folder called 'common', and hence no mongoDB folder either.  (Only 'include', 'launch', & 'src'. There is a CMakeLists.txt & package.xml too.)

So I created the common/mongoDB folders and ran it again, and this time a new error showed:

    Error [SDF.cc:788] Missing element description for [pointCloudCutoffMax]

This time, after running the script, it seemed to work (at least most of it).  I had the way points and paths showing in Gazebo. rqt showed output in the 2nd, 3rd, & 4th panes but the 1st pane was empty.  In the terminal, where I did the launch, the output showed an error in the PDDL plan:

    [ INFO] [1444147293.767550506, 514.970000000]: KCL: (PS) Running: timeout 10 /home/user/Documents/thesis/ROSPlan/catkin_ws/src/rosplan/rosplan_planning_system/common/bin/popf -n /home/user/Documents/thesis/ROSPlan/catkin_ws/src/rosplan/rosplan_planning_system/common/domain.pddl /home/user/Documents/thesis/ROSPlan/catkin_ws/src/rosplan/rosplan_planning_system/common/problem.pddl > /home/user/Documents/thesis/ROSPlan/catkin_ws/src/rosplan/rosplan_planning_system/common/plan.pddl

I will analyse the error tomorrow.

Regards,
Nap.

m312z, 2015-10-07:
Hello,

Thanks for the analysis so far. That's pretty thorough.

The script should not have to be run from the rosplan folder (it only contains rosservice calls).

Github does not remember empty folders. I've put an empty file in "common/mongoDB" so nobody else should get this problem.

`Error [SDF.cc:788] Missing element description for [pointCloudCutoffMax]`
I don't recognise this error. I believe it is coming from Gazebo.

The final line:
```
timeout 10 .../popf -n .../domain.pddl .../problem.pddl > .../plan.pddl
```
is the planner. The planner we use is POPF. We run it with a timeout of 10 seconds; the "-n" argument means that it tries to improve the first plan until the timeout. You can change this command in the launch file. If the command is not working you can copy/paste it and try it alone to diagnose the error.

Once a plan has been generated, it will appear in the 1st pane.

Michael

```

```

m312z, 2015-11-06:
Hello,

Did you find the problem?

If you are still waiting for the planner to complete after 10 seconds, my guess is that `timeout 10 ...` was not killing the planning process.

You can test this by copy and pasting the command, and trying it alone. The process should be automatically killed after 10 seconds. If not, then timeout is not working properly.

You could get around this for now by removing timeout and the argument `-n` from the command line in the launch file. This means POPF will return the first plan found, rather than attempt to improve the plan for 10 seconds.

I will close this issue now.

Michael

closed 2015-11-06 by m312z

Napoleon-BlownApart, 2015-11-06:
Sorry Michael, I've been pressed with other things I've been doing. I will come back to this in a few days, but I think you're right in closing the issue.  What needed to happen did.
I'm glad the feedback was useful for you as it will make life easier for others downstream.

If I have any further comments, I will post later.

Thanks,
Nap

m312z, 2015-11-06:
Hello,

No problem. We're still working on ROSPlan to introduce new features, and of course new bugs. Comments like this are always useful.

I you have any further problems, open another issue, so that it will be easier for others to find.

Michael

ashishpatel1992, 2020-04-19:
I am using melodic and I solved the issue using
`sudo apt-get install ros-melodic-navigation`

> [Ubuntu 14.04.1 fully updated, ROS Indigo]
>
> I followed the instructions but when I built using 'catkin_make' I got an error informing me that I didn't have GLUT. So I installed freeglut3 (which [...]