KCL-Planning/ROSPlan #12: cant find "amcl_mc2.launch"

mcqrobot, 2015-11-06:
Hi!
   I am a  student and when I first see this package i was so excited because it is so newest and amazing idea. now i show my problem.
  In the file "rosplan_demos/launch/mc2_turtlebot.launch", it need bring up the real robot--turtlebot, and need launch the "turtlebot_navigation/launch/amcl_mc2.launch" ,however, i cannot locate and find the launch file named "amcl_mc2.launch" in my workspace OR "/opt/ros/indigo/share..."path.   so when I run this lanuch file, ROS show the error that "No such file or ....“ what should i do ,i can not continue my reasearch. can you help me?

m312z, 2015-11-06:
Hello,

You will need to modify "mc2_turtlebot.launch" to change one line:
`<include file="$(find turtlebot_navigation)/launch/amcl_mc2.launch" >`
This refers to a file you do not have.

"amcl_mc2.launch" is a modified version of "amcl_demo.launch" from turtlebot_navigation. The only difference is the map file name. 

What you do next depends on whether you are using ROS Hydro or Indigo.

*Hydro*

You could create your own copy of the amcl_demo file, with a reference to the real map that you wish to use:
https://github.com/turtlebot/turtlebot_apps/blob/hydro/turtlebot_navigation/launch/amcl_demo.launch
(line 13)

*Indigo*

In Indigo you can see that the map file name is now an environment variable.

m312z, 2015-11-06:
As an example, here is the contents of amcl_mc2.launch:
```
<launch>
  <include file="$(find turtlebot_bringup)/launch/3dsensor.launch">
    <arg name="rgb_processing" value="false" />
    <arg name="depth_registration" value="false" />
    <arg name="depth_processing" value="false" />
    
    <!-- We must specify an absolute topic name because if not it will be prefixed by "$(arg camera)".
         Probably is a bug in the nodelet manager: https://github.com/ros/nodelet_core/issues/7 --> 
    <arg name="scan_topic" value="/scan" />
  </include>

  <!-- Map server -->
  <arg name="map_file" default="$(find turtlebot_navigation)/maps/map_studies.yaml"/>
  <node name="map_server" pkg="map_server" type="map_server" args="$(arg map_file)" />

  <arg name="initial_pose_x" default="0.0"/> <!-- Use 17.0 for willow's map in simulation -->
  <arg name="initial_pose_y" default="0.0"/> <!-- Use 17.0 for willow's map in simulation -->
  <arg name="initial_pose_a" default="0.0"/>
  <include file="$(find turtlebot_navigation)/launch/includes/amcl.launch.xml">
    <arg name="initial_pose_x" value="$(arg initial_pose_x)"/>
    <arg name="initial_pose_y" value="$(arg initial_pose_y)"/>
    <arg name="initial_pose_a" value="$(arg initial_pose_a)"/>
  </include>

  <include file="$(find turtlebot_navigation)/launch/includes/move_base.launch.xml"/>

</launch>
```

closed 2015-11-06 by m312z

mcqrobot, 2015-11-06:
thanks a lot. i will try according your advise.