KCL-Planning/ROSPlan #13: the planner problem

mcqrobot, 2015-12-16:
hi Michael  and etc:
    your ros packages"ROSPlan" released is really a greate work, and  I have using it for several weeks and I found the planner of the planning system is POPF, it is just an executable file, but I want to see the code of the planner. Besides , If I want to change another planner, What should I ajust and modify the package? If you can share the code of planners, can you tell me or send the email to me:mcq_work@yahoo.com
  I am keeping your reply, Thanks a lot.

m312z, 2015-12-16:
Hello,

Thanks for your interest! 

More information on POPF can be found here:
http://www.inf.kcl.ac.uk/research/groups/PLANNING/index.php?option=com_content&view=article&id=46&Itemid=55

And the source code can be found on the sourceforge page here:
http://sourceforge.net/projects/tsgp/

To use another planner, you must do two things:
1. Alter the command line for the planner, which is found in the launch file:
```
<node name="rosplan_planning_system" pkg="rosplan_planning_system" type="planner" respawn="false" output="screen">
		...
		<!-- to run the planner -->
		<param name="planner_command" value="timeout 10 $(find rosplan_planning_system)/common/bin/popf -n" />
</node>
```
The command line will then be:
timeout 10 $(find rosplan_planning_system)/common/bin/popf -n <domain> <problem>

If you need to change it further, the code can be found in `rosplan_planning_system/src/PlanningSystem.cpp` (lines 193-199)

2. If the output is different from POPF output, then you must alter the code that parses the plan.
This can be found in `rosplan_planning_system/src/PlanParser.cpp` This code reads in the plan from file, and generates a list of action messages to be sent.

Michael

closed 2015-12-16 by m312z

mcqrobot, 2015-12-17:
Hi,
    Thanks for your reply. 
     Actually,the POPF page "http://www.inf.kcl.ac.uk/research/groups/PLANNING/index.php?option=com_content&view=article&id=46&Itemid=55 "

   and source download page" http://sourceforge.net/projects/tsgp/" 
   has been tried by me several days ago, but unfortunately, when I follow its step and run the command "./build"　to build the popf2(filename is tempo-sat-popf2), it throw the error that some variables are not found". When I try to run the command "./run-cmake-release" to compile  OPTIC (filename is optic),but it also throw the error. All in all, I guess that the things such as "popf2-clp","popf3-clp","optic-clp" are needed but I don't have.
    
    Er...　I don't know whether you understand what I said. Anyhow, thanks a lot for your help.

Neowizard, 2015-12-17:
Just to point out, you can actually replace the planner used buy RoSPlan.
It requires you to rewrite the parser that reads the planner results (I don't remember which file it was, but it wasn't hard to find). Other than that, IIRC, it's just a matter of changing the command line that invokes the planner in the launch file.

m312z, 2015-12-17:
POPF is a separate tool from ROSPlan -- your best bet is to contact its developer, Andrew Coles:
http://www.inf.kcl.ac.uk/staff/andrew/

mcqrobot, 2015-12-17:
Thank you so much! ~_~

mcqrobot, 2015-12-18:
Thanks for your advice and I used annother planner well support PDDL scuccessfully according to your advice. I changed the param value in lauch file and modified the file "PlanPaser.cpp" and now I can use annother planner.  Thanks very much!