KCL-Planning/ROSPlan #16: After install the packages the headers can not be found

ipa-nhg, 2016-02-23:
Some header  files are not being installed (rosplan_planning_system/src/VALfiles/*.h) , they are not defined in the CMakeLists

Also you should not use the relative path to find the headers:

- https://github.com/KCL-Planning/ROSPlan/blob/master/rosplan_planning_system/include/rosplan_planning_system/PlanningEnvironment.h#L9

after install the package you will not be able to find the folder ../../src , it will not exist. A simple solution can be just move all the header files into the include/ folder .

ipa-nhg, 2016-02-23:
A possible solution :https://github.com/KCL-Planning/ROSPlan/pull/17

m312z, 2016-04-07:
#17 is merged, is this solved?

closed 2016-05-19 by m312z