KCL-Planning/ROSPlan #18: information store problem

mcqrobot, 2016-03-11:
hi:
   I have tested this project for weeks, and  I feel the information such as "instances","facts" are just stored in STL and obtained from STL when it used. I don't know the mongodb whether is used. and I don't see the codes that related the mongodb. Because, if you want to operate the Mongodb ,you should write a client code, but I can not find it in this project. Can you tell me where it. THX!

m312z, 2016-03-11:
Hello,
Instances and facts (and in general the PDDL model) are stored in the Knowledge Base, but not in the MongoDB.
The MongoDB is used to store the real information that the PDDL model represents. For example:
1. a Waypoint object is stored in the KB, as part of the PDDL model.
2. a geometry_msgs::PoseStamped message is stored in the MongoDB with a key corresponding to the PDDL object name.

Hope this helps,
Michael

mcqrobot, 2016-03-12:
Hi: 
   I got it. However, I find that the MongoDB is only in PRM roadmap setup, to store the real waypoint information. If  I just want to use the "RPSimpleMapServer.cpp", which don't need the PRM to create the roadmap,  in this situation, whether means I don't need the Mongodb and I can delete the following in launch file "interfaced_planning_system.launch":
	
	<node name="rosplan_scene_database" pkg="mongodb_store" type="mongodb_server.py" respawn="false" output="log">
	    <param name="database_path" value="$(find rosplan_knowledge_base)/common/mongoDB" />
	</node>
	<node name="rosplan_scene_message_store" pkg="mongodb_store" type="message_store_node.py" respawn="false" output="log">
	</node>

 So then Knowledge Base don't need the MongoDB at all, is it right?

----by  mark

m312z, 2016-03-12:
Hello

RPSimpleRoadmap reads waypoint data from file instead of the PRM. (I prefer to use this, as my map does not change.) However, the waypoint data is still stored in the MongoDB (check RPSimpleMapServer.cpp line 189).

The real coordinates must be stored, otherwise when it is time to dispatch a goto_waypoint action, the robot will not have a real navigation goal.

Michael

mcqrobot, 2016-03-12:
Thanks a lot!
     I have checked the code carefully again and I have solved this problem!

Mark

m312z, 2016-03-12:
Happy to help!
Michael

closed 2016-03-12 by m312z

yashu88, 2018-02-14:
In case of ROSPlan
1) information such as "instances","facts" are stored in Standard Template Library and obtained from STL? or 
2)knowledge management service with the help of .srv and .msg file just import and export information between nodes? 

Then what is the use of MongoDB here? Is problem generation or plan dispatch file uses message_store.0 file contents in decision making?

Kindly help me out on this.

dgerod, 2018-02-15:
See my comment in the other issue.