KCL-Planning/ROSPlan #20: Turtlebot demo starts, but nothing happens

thrift, 2016-04-01:
I'm trying to get ROSPlan going on Ubuntu 14.04.4 LTS and ROS Indigo as per standard install instructions. Everything seems to have worked fine and the demo starts with Gazebo and rviz windows looking as expected. But the robot doesn't move and I don't get the expected output in the terminal. Here's what I get (funny characters are due to Emacs shell), and nothing happens after:

   smbdy@deep:~$ source ~/workplace/disser/devel/ROSPlan/devel/setup.sh
   smbdy@deep:~$ roslaunch rosplan_demos turtlebot.launch
   ... logging to /home/smbdy/.ros/log/3aafb094-f800-11e5-ac35-a088b4c0f0e4/roslaunch-deep-7721.log
   Checking log directory for disk usage. This may take awhile.
   Press Ctrl-C to interrupt
   Done checking log file disk usage. Usage is <1GB.
   ]2;/home/smbdy/workplace/disser/devel/ROSPlan/src/rosplan/rosplan_demos/launch/turtlebot.launch
   started roslaunch server http://deep:42850/

   SUMMARY
   ========

   PARAMETERS
    * /amcl/base_frame_id: base_footprint
    * /amcl/global_frame_id: map
    * /amcl/gui_publish_rate: 10.0
    * /amcl/initial_pose_a: 0.0
    * /amcl/initial_pose_x: 0.0
    * /amcl/initial_pose_y: 0.0
    * /amcl/kld_err: 0.05
    * /amcl/kld_z: 0.99
    * /amcl/laser_lambda_short: 0.1
    * /amcl/laser_likelihood_max_dist: 2.0
    * /amcl/laser_max_beams: 60
    * /amcl/laser_max_range: 12.0
    * /amcl/laser_model_type: likelihood_field
    * /amcl/laser_sigma_hit: 0.2
    * /amcl/laser_z_hit: 0.5
    * /amcl/laser_z_max: 0.05
    * /amcl/laser_z_rand: 0.5
    * /amcl/laser_z_short: 0.05
    * /amcl/max_particles: 2000
    * /amcl/min_particles: 500
    * /amcl/odom_alpha1: 0.2
    * /amcl/odom_alpha2: 0.2
    * /amcl/odom_alpha3: 0.2
    * /amcl/odom_alpha4: 0.2
    * /amcl/odom_alpha5: 0.1
    * /amcl/odom_frame_id: odom
    * /amcl/odom_model_type: diff
    * /amcl/recovery_alpha_fast: 0.0
    * /amcl/recovery_alpha_slow: 0.0
    * /amcl/resample_interval: 1
    * /amcl/transform_tolerance: 1.0
    * /amcl/update_min_a: 0.2
    * /amcl/update_min_d: 0.25
    * /amcl/use_map_topic: False
    * /bumper2pointcloud/pointcloud_radius: 0.24
    * /cmd_vel_mux/yaml_cfg_file: /opt/ros/indigo/s...
    * /depthimage_to_laserscan/output_frame_id: /camera_depth_frame
    * /depthimage_to_laserscan/range_min: 0.45
    * /depthimage_to_laserscan/scan_height: 10
    * /domain_path: /home/smbdy/workp...
    * /move_base/DWAPlannerROS/acc_lim_theta: 2.0
    * /move_base/DWAPlannerROS/acc_lim_x: 1.0
    * /move_base/DWAPlannerROS/acc_lim_y: 0.0
    * /move_base/DWAPlannerROS/forward_point_distance: 0.325
    * /move_base/DWAPlannerROS/global_frame_id: odom
    * /move_base/DWAPlannerROS/goal_distance_bias: 24.0
    * /move_base/DWAPlannerROS/max_rot_vel: 5.0
    * /move_base/DWAPlannerROS/max_scaling_factor: 0.2
    * /move_base/DWAPlannerROS/max_trans_vel: 0.5
    * /move_base/DWAPlannerROS/max_vel_x: 0.5
    * /move_base/DWAPlannerROS/max_vel_y: 0.0
    * /move_base/DWAPlannerROS/min_rot_vel: 0.4
    * /move_base/DWAPlannerROS/min_trans_vel: 0.1
    * /move_base/DWAPlannerROS/min_vel_x: 0.0
    * /move_base/DWAPlannerROS/min_vel_y: 0.0
[... 247 lines cut ...]
   [ INFO] [1459511562.317000318, 0.200000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
   [ INFO] [1459511562.318702736, 0.200000000]: Kobuki(ns = //): Advertise Cliff[mobile_base/events/cliff]!
   [ INFO] [1459511562.321078593, 0.200000000]: Kobuki(ns = //): Advertise Bumper[mobile_base/events/bumper]!
   [ INFO] [1459511562.322978787, 0.200000000]: Kobuki(ns = //): Advertise IMU[mobile_base/sensors/imu_data]!
   [ INFO] [1459511562.323155618, 0.200000000]: GazeboRosKobuki plugin ready to go! [mobile_base]
   [spawn_turtlebot_model-4] process has finished cleanly
   log file: /home/smbdy/.ros/log/3aafb094-f800-11e5-ac35-a088b4c0f0e4/spawn_turtlebot_model-4*.log
   [ INFO] [1459511564.032256007, 1.900000000]: Using plugin "static_layer"
   [ INFO] [1459511564.134177786, 2.000000000]: Requesting the map...
   [ INFO] [1459511564.354617357, 2.220000000]: Resizing costmap to 231 X 194 at 0.050000 m/pix
   [ INFO] [1459511564.453515783, 2.320000000]: Received a 231 X 194 map at 0.050000 m/pix
   [ INFO] [1459511564.459569563, 2.320000000]: Using plugin "obstacle_layer"
   [ INFO] [1459511564.462421890, 2.320000000]:     Subscribed to Topics: scan bump
   [ INFO] [1459511564.578152905, 2.440000000]: Using plugin "inflation_layer"
   [ INFO] [1459511564.925718098, 2.770000000]: Using plugin "obstacle_layer"
   [ INFO] [1459511565.019982478, 2.860000000]:     Subscribed to Topics: scan bump
   [ INFO] [1459511565.189085933, 3.030000000]: Using plugin "inflation_layer"
   [ INFO] [1459511565.433356034, 3.280000000]: Created local_planner dwa_local_planner/DWAPlannerROS
   [ INFO] [1459511565.438562160, 3.280000000]: Sim period is set to 0.20
   [ INFO] [1459511566.023279987, 3.870000000]: Recovery behavior will clear layer obstacles
   [ INFO] [1459511566.124972369, 3.970000000]: Recovery behavior will clear layer obstacles
   [ INFO] [1459511566.232671290, 4.070000000]: odom received!
   [ INFO] [1459511566.374772713, 4.210000000]: KCL: (MoveBase) Ready to receive
[image]
[image]

m312z, 2016-04-01:
Hello,

It does look like everything has started up correctly. I see: "ready to receive" for the planning system (PS) and Knowledge Management System (KB). But there are no goals, and no request to plan yet.

Have you remembered to run the start script?
```
sh src/rosplan_demos/scripts/turtlebot_explore.bash 
```

turtlebot_explore.bash does:
- generates a roadmap (you will see in rviz);
- adds the planning goals; and
- calls the planning system service.

thrift, 2016-04-01:
Thanks, Michael. Yes, it was the script, I didn't realise it needs to be run every time. But now I have another problem: the planner seems to be failing.

This is script output:
    smbdy@deep:~/workplace/disser/devel/ROSPlan$ src/rosplan/rosplan_demos/scripts/turtlebot_explore.bash
    waypoints: ['wp0', 'wp1', 'wp2', 'wp3', 'wp4']
    success: True
    success: True
    success: True
    success: True
    success: True
    success: True
    success: True
    success: True
    success: True
    success: True
    success: True

    smbdy@deep:~/workplace/disser/devel/ROSPlan$ 

This is terminal output (in addition to what I already supplied):

    [ INFO] [1459525780.041406125, 243.070000000]: KCL: (RPRoadmapServer) Cleaning old roadmap
    [ INFO] [1459525780.045249880, 243.070000000]: KCL: (RPRoadmapServer) Generating roadmap
    [ INFO] [1459525780.046240370, 243.070000000]: KCL: (RPRoadmapServer) Adding knowledge
    [ INFO] [1459525780.050205850, 243.080000000]: KCL: (KB) Adding instance (waypoint, wp0)
    [ INFO] [1459525780.056264528, 243.090000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.060652570, 243.090000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.067273263, 243.090000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.076331605, 243.090000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.082513169, 243.090000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.096072555, 243.100000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.101786594, 243.100000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.106422056, 243.110000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.140312865, 243.140000000]: KCL: (KB) Adding instance (waypoint, wp1)
    [ INFO] [1459525780.145482499, 243.140000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.150278643, 243.150000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.154750075, 243.150000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.158368519, 243.150000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.163833473, 243.150000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.174788248, 243.160000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.182992168, 243.170000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.192684106, 243.170000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.201842522, 243.180000000]: KCL: (KB) Adding instance (waypoint, wp2)
    [ INFO] [1459525780.206175185, 243.180000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.210098797, 243.190000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.218143273, 243.200000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.224714484, 243.200000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.229539111, 243.210000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.233384220, 243.210000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.238668874, 243.220000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.242974234, 243.220000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.264940028, 243.240000000]: KCL: (KB) Adding instance (waypoint, wp3)
    [ INFO] [1459525780.269107247, 243.240000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.273827466, 243.240000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.277858555, 243.240000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.287352517, 243.250000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.296063006, 243.250000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.303509587, 243.270000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.307846730, 243.280000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.310328813, 243.280000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.319040036, 243.290000000]: KCL: (KB) Adding instance (waypoint, wp4)
    [ INFO] [1459525780.321695096, 243.290000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.324659083, 243.300000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.330983321, 243.300000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.335147239, 243.300000000]: KCL: (KB) Adding domain attribute (connected)
    [ INFO] [1459525780.341227385, 243.310000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.346445413, 243.320000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.352490457, 243.320000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.356181752, 243.330000000]: KCL: (KB) Adding domain function (distance)
    [ INFO] [1459525780.364901571, 243.340000000]: KCL: (KB) Adding domain attribute (robot_at)
[... 122 lines cut ...]
        (visited wp3)
        (visited wp4)
        (visited wp5)
        (visited wp6)
        (visited wp7)
        (visited wp8)
[... cut ...]
    /home/smbdy/workplace/disser/devel/ROSPlan/src/rosplan/rosplan_planning_system/common/problem.pddl: line: 58: Warning: Undeclared symbol: wp7
    /home/smbdy/workplace/disser/devel/ROSPlan/src/rosplan/rosplan_planning_system/common/problem.pddl: line: 59: Warning: Undeclared symbol: wp8
    /home/smbdy/workplace/disser/devel/ROSPlan/src/rosplan/rosplan_planning_system/common/problem.pddl: line: 60: Warning: Undeclared symbol: wp9
    Object with unknown type: wp5
    smbdy@deep:~/workplace/disser/devel/ROSPlan$ 

RQT screenshot:

[image]

Thanks in advance.

m312z, 2016-04-01:
Hello,

I see the problem. In the PDDL problem file there are some goals that refer to objects that don't exist. This is a problem with the script I have been meaning to fix for some time...

The waypoint generation generates *up to* 9 waypoints, but sometimes less. In your problem file there are 5 waypoints (wp0 -- wp4).

However, the script always adds goals to visit 10 waypoints (wp0 -- wp9).

If you edit the script to add fewer goals it will be more robust.

Alternatively, simply remove the extra goals (or add the goals you want) using the rqt interface and press the plan button to plan and execute again.

Michael

thrift, 2016-04-01:
Thanks Michael. Increasing maximum number of attempts in waypoint generation call solved it. Everything seems to work though the robot is making a huge number of 360 degree turns for no apparent reason.  Shall I push the script fix into the repo?

Egor.

closed 2016-04-01 by thrift

m312z, 2016-04-01:
Feel free to make a pull request. However, I think the best fix is either ensuring goals are only added for real waypoints, or to use the simpleRoadmap that reads waypoints from file.