KCL-Planning/ROSPlan #21: New files

mcqrobot, 2016-04-06:
Hi:
   I noticed that in package "rosplan_planning_system", some new cpp files appeared in "src" folder. 

1 EsterelPlanDispatcher.cpp
  I don't know what it is. In its constructor, it loads:

 	EsterelPlanDispatcher::EsterelPlanDispatcher() {
		ros::NodeHandle nh("~");
		nh.param("strl_file_path", strl_file, std::string("common/plan.strl"));
	}

What is the "plan.strl" file and how can genereate the strl file?

2 CFFPlanParser.cpp

   I think it is about Contingent-FF planner, so I downloaded  an executable Contingent-FF from here: https://fai.cs.uni-saarland.de/hoffmann/cff.html

   However I don't know whether it well support PDDL. So before I used it I test it like this:

   mcq@mcq-ubuntu:~/ROSPlan/src/rosplan/rosplan_planning_system/common$ ./bin/Contingent-FF -p domain.pddl 
  ff: parsing domain file
  domain.pddl: syntax error in line 3, ':FLUENTS':
 domain definition expected
 ff: parsing domain file
 domain.pddl: syntax error in line 3, ':DURATIVE-ACTIONS':
 domain definition expected

   It seems that the planner can not support FLUENTS and DURATIVE-ACTIONS and etc. I don't know how to correctly use it.

3 As other new files such as "POPFPlanParser.cpp" and "SimplePlanDispatcher.cpp"  are easy to understand and use.

Mark

m312z, 2016-04-06:
Hello,

Yes there are a lot more files. We are making a number of improvements and adding new functionality. (Most of the work is in the other branch "squirrel").

1&2. CFFPlanParser is for parsing Contingent-FF plans. This will only work on non-temporal domains. EsterelPlanDispatcher is used for dispatching these plans. The versions of these in the master branch might not work correctly.

3. POPFPlanParser and SimplePlanDispatcher: these are the same as before, and should be the default setting in the master branch.

There is still some way to go until things are stable enough to merge back into the master branch. However, then things should become much more robust.
- the dispatcher properly waits for action preconditions to be stisfied before dispatching an action.
- it also can deal with contingent (branching) plans.
- the plans are output dynamically in DOT format, and can be visualised in a browser as they execute (using the d3 library.)
- the planning system can be started as a ROS action as well as service. This means that a planning system can be spawned and started as an action in another plan.

For now, as long as the master branch remains stable, these extra files do nothing.

Michael

mcqrobot, 2016-04-06:
I see. Thx!

closed 2016-05-19 by m312z