KCL-Planning/ROSPlan #22: Help setting Waypoints

DanniloMiranda, 2016-04-07:
Hi.

I'd like the robot to visit certain places, so I changed the file "waypoints.txt" with the points that i what. I changed the file "turtlebot_explore.bash" as shown below

#!/bin/bash
rosservice call /kcl_rosplan/update_knowledge_base "update_type: 0
knowledge:
  knowledge_type: 0
  instance_type: 'robot'
  instance_name: 'kenny'
  attribute_name: ''
  function_value: 0.0";
for i in 0 1 2 
do
rosservice call /kcl_rosplan/update_knowledge_base "update_type: 1
knowledge:
  knowledge_type: 1
  instance_type: ''
  instance_name: ''
  attribute_name: 'visited'
  values:
  - {key: 'wp', value: 'wp$i'}
  function_value: 0.0"
done;
rosservice call /kcl_rosplan/planning_server;
-----------------------------------------------------

And I get this result:

[ INFO] [1460025104.625814104, 293.070000000]: KCL: (KB) Adding instance (robot, kenny)
[ INFO] [1460025105.765086602, 294.200000000]: KCL: (KB) Adding mission goal (visited)
[ INFO] [1460025106.447521278, 294.830000000]: KCL: (KB) Adding mission goal (visited)
[ INFO] [1460025107.654359569, 295.990000000]: KCL: (KB) Adding mission goal (visited)
[ INFO] [1460025108.903157257, 297.170000000]: KCL: (PS) Parsing domain: /home/dannilo/catkin_ws/src/ROSPlan/src/ROSPlan/rosplan_planning_system/common/domain.pddl.
[ INFO] [1460025108.904104001, 297.170000000]: KCL: (PS) Fetching objects
[ INFO] [1460025108.922503007, 297.200000000]: KCL: (PS) Generating PDDL problem file
[ INFO] [1460025108.924013917, 297.200000000]: KCL: (PS) Running: timeout 10 /home/dannilo/catkin_ws/src/ROSPlan/src/ROSPlan/rosplan_planning_system/common/bin/popf -n /home/dannilo/catkin_ws/src/ROSPlan/src/ROSPlan/rosplan_planning_system/common/domain.pddl /home/dannilo/catkin_ws/src/ROSPlan/src/ROSPlan/rosplan_planning_system/common/problem.pddl > /home/dannilo/catkin_ws/src/ROSPlan/src/ROSPlan/rosplan_planning_system/common/plan.pddl
-----------------------------------------------------

The supplied domain/problem file appear to violate part of the PDDL
language specification.  Specifically:

The planner will continue, but you may wish to fix your files accordingly
Type Errors Encountered in Problem File
---------------------------------------

Due to type errors in the supplied problem file, the planner
has to terminate.  The log of type checking is as follows:

[ INFO] [1460025109.335038032, 297.570000000]: KCL: (PS) Planning complete
[ INFO] [1460025109.335172337, 297.570000000]: KCL: (PS) Plan was unsolvable! Try again?
[ INFO] [1460025109.335564494, 297.570000000]: KCL: (PS) Dispatching plan
[ INFO] [1460025109.337285493, 297.570000000]: KCL: (PS) Planning System Finished

---------------------------------------

The Domain is the same as example. What else I have to change?

Dannilo

m312z, 2016-04-08:
Hello,

You might have to change the waypoint server being launched. There are two types: the first generates waypoints (rproadmapServer), the other loads waypoints from a file (rpsimplemapServer).

The default launch file includes rproadmapServer. This is then called in the first line of the original bash script to generate the waypoints.

I believe your error is that since this line is removed, there are no waypoints at all, and the goals refer to missing waypoints. To check this you can:
1. Look at the problem file "problem.pddl" and see if the waypoints are there.
2. Start rqt interface and see the objects window (see the wiki here: https://github.com/KCL-Planning/ROSPlan/wiki/d.-User-Interface)
3. Run the command line you pasted above to see the output from the planner directly.

To instead read waypoints from a file you need to alter the launch file to load the rpsimplemapServer.

(The launch files are a bit over-complicated right now, but it is an easy change)

"rosplan_demos/launch/turtlebot.launch"
...includes...
"rosplan_planning_system/launch/interfaced_planning_system.launch"
...which includes
..."rosplan_interface_mapping/launch/rosplan_roadmap_server.launch"

In this last file you can change "rproadmapServer" to "rpsimplemapServer".

Please take a look at the launch file mc2_turtlebot.launch for more details. It uses the rpsimplemapServer like this:
```
	<param name="waypoint_file" value="$(find rosplan_demos)/scripts/waypoint_file.txt" />

	<node name="rosplan_roadmap_server" pkg="rosplan_interface_mapping" type="rpsimplemapServer" respawn="false" output="screen">
	    <param name="data_path" value="$(find rosplan_planning_system)/common/" />
	</node>
```

DanniloMiranda, 2016-04-10:
Before all, I would like to thank you for the aswer, it was really helpfull. 

 I checked the problem.pdll file and noticed that the goals are really refering to missing waypoints.
  
  I have lack of experience with ROS. What I did was to add the following lines to the begining of turtlebot_explore.bash":

rosservice call /kcl_rosplan/roadmap_server/create_prm "{nr_waypoints: 3, min_distance: 0.3, casting_distance: 2.0, connecting_distance: 8.0, occupancy_threshold: 10, total_attempts: 10}";

Besides that, I changed. the file "rosplan_interface_mapping/launch/rosplan_roadmap_server.launch" to: 

 <launch>
	
	 <param name="waypoint_file" value="$(find rosplan_demos)/scripts/waypoint_file.txt" />

    <node name="rosplan_roadmap_server" pkg="rosplan_interface_mapping" type="rpsimplemapServer" respawn="false" output="screen">
        <param name="data_path" value="$(find rosplan_planning_system)/common/" />
    </node>
</launch>

Then I got the follwing output:

[ INFO] [1460233221.246686035, 1853.200000000]: KCL: (RPSimpleMapServer) Cleaning old roadmap
[ INFO] [1460233221.248126485, 1853.200000000]: KCL: (KB) Removing instance (waypoint, ALL)
[ INFO] [1460233221.248169811, 1853.200000000]: KCL: (KB) Removing instance (waypoint, ALL)
[ INFO] [1460233221.261593314, 1853.210000000]: KCL: (RPSimpleMapServer) Generating roadmap
[ INFO] [1460233221.261693580, 1853.210000000]: KCL: (RPSimpleMapServer) Adding knowledge
[ INFO] [1460233221.264400109, 1853.220000000]: KCL: (KB) Adding instance (waypoint, wp0)
[ INFO] [1460233221.276626816, 1853.230000000]: KCL: (KB) Adding instance (waypoint, wp2)
[ INFO] [1460233221.280469986, 1853.230000000]: KCL: (RPSimpleMapServer) Done
[ INFO] [1460233225.803945376, 1857.650000000]: KCL: (PS) Fetching objects
[ INFO] [1460233225.819927906, 1857.660000000]: KCL: (PS) Generating PDDL problem file
[ INFO] [1460233225.820262857, 1857.660000000]: KCL: (PS) Running: timeout 10 /home/dannilo/catkin_ws/src/ROSPlan/src/ROSPlan/rosplan_planning_system/common/bin/popf -n /home/dannilo/catkin_ws/src/ROSPlan/src/ROSPlan/rosplan_planning_system/common/domain.pddl /home/dannilo/catkin_ws/src/ROSPlan/src/ROSPlan/rosplan_planning_system/common/problem.pddl > /home/dannilo/catkin_ws/src/ROSPlan/src/ROSPlan/rosplan_planning_system/common/plan.pddl
[ INFO] [1460233225.825444152, 1857.660000000]: KCL: (PS) Planning complete
[ INFO] [1460233225.825986024, 1857.660000000]: KCL: (PS) Clean and update knowledge filter
[ INFO] [1460233225.826241502, 1857.660000000]: KCL: (PS) Dispatching plan
[ INFO] [1460233225.826297578, 1857.660000000]: KCL: (PS) Planning System Finished

  After that I checked the problem.dll file:

(define (problem turtlebot_demo_task)
(:domain turtlebot_demo)
(:objects
    kenny - robot
    wp0 wp1 wp2 - waypoint
)
(:init
)
(:goal (and
    (visited wp0)
    (visited wp1)
    (visited wp2)
)))

And the generated plan was:

Number of literals: 6
Constructing lookup tables:
Post filtering unreachable actions: 
[01;34mNo analytic limits found, not considering limit effects of goal-only operators[00m
Pruning (goto_waypoint kenny wp0 wp0) - never appeared in initial RPG
Pruning (goto_waypoint kenny wp1 wp0) - never appeared in initial RPG
Pruning (goto_waypoint kenny wp2 wp0) - never appeared in initial RPG
Pruning (goto_waypoint kenny wp0 wp1) - never appeared in initial RPG
Pruning (goto_waypoint kenny wp1 wp1) - never appeared in initial RPG
Pruning (goto_waypoint kenny wp2 wp1) - never appeared in initial RPG
Pruning (goto_waypoint kenny wp0 wp2) - never appeared in initial RPG
Pruning (goto_waypoint kenny wp1 wp2) - never appeared in initial RPG
Pruning (goto_waypoint kenny wp2 wp2) - never appeared in initial RPG
Initial heuristic = -1.000
;; Problem unsolvable!
; Time 0.00

How can I generate the initial states? 
Thank you again for the attention

m312z, 2016-04-10:
Hello,

1) The change you have made to the waypoint launch server is good. The new node should be loading the waypoints described in the file:
rosplan_demos/scripts/waypoint_file.txt

You will want to ensure that these are the fixed waypoints that you require.

2) In the Bash script, you don't need the line:
```
rosservice call /kcl_rosplan/roadmap_server/create_prm "{nr_waypoints: 3, min_distance: 0.3, casting_distance: 2.0, connecting_distance: 8.0, occupancy_threshold: 10, total_attempts: 10}";
```
This line calls the Probabilistic roadmap generator that you have replaced. Since you are loading waypoints from the file, then you don't require this line.

3) you're right -- the initial state is not set! I should have included this in the turtlebot_explore.bash.

We will generate this in the script. After the lines which add the robot instance.
```
rosservice call /kcl_rosplan/update_knowledge_base "update_type: 0
knowledge:
knowledge_type: 1
instance_type: ''
instance_name: ''
attribute_name: 'robot_at'
  values:
  - {key: 'v', value: 'kenny'}
  - {key: 'wp', value: 'wp0'}
function_value: 0.0";
```

This will add the fact: (robot_at kenny wp0) to the current state.

I'll update the demo script in the repository for this too.

4) You can vew all of the current objects and facts using the user interface. You can start this by:
source devel/setup.bash
rqt --standalone rosplan_rqt

It should look like the screenshot on the wiki here:
https://github.com/KCL-Planning/ROSPlan/wiki/d.-User-Interface

DanniloMiranda, 2016-04-13:
It solved my problem.

Thank you so much!!

closed 2016-05-18 by DanniloMiranda