KCL-Planning/ROSPlan #24: a. Installation issue

MahmoudElShazly, 2016-04-10:
I am trying to install KCL-Planning, followed the steps here:
https://github.com/KCL-Planning/ROSPlan/wiki/a.-Installation

and I got this errors, it says it cannot find a header file, is there an issue in the packages dependencies?

[  0%] [  0%] Built target _rosplan_dispatch_msgs_generate_messages_check_deps_ActionDispatch
Built target _rosplan_dispatch_msgs_generate_messages_check_deps_ActionFeedback
[  0%] Built target diagnostic_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target _rosplan_dispatch_msgs_generate_messages_check_deps_CompletePlan
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target diagnostic_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target diagnostic_msgs_generate_messages_py
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_Filter
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_KnowledgeUpdateService
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_GetDomainAttributeService
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_GetAttributeService
[  0%] [  0%] Built target geometry_msgs_generate_messages_py
Built target _rosplan_knowledge_msgs_generate_messages_check_deps_DomainFormula
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_GetInstanceService
[  0%] [  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_AddWaypoint
Built target _rosplan_knowledge_msgs_generate_messages_check_deps_Notification
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_GetDomainOperatorService
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_CreatePRM
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_GetDomainTypeService
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_KnowledgeQueryService
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_KnowledgeItem
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] [  0%] Built target std_srvs_generate_messages_cpp
Built target roscpp_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target std_srvs_generate_messages_py
[  0%] Built target std_srvs_generate_messages_lisp
[  0%] Built target mongodb_store_msgs_generate_messages_cpp
[  0%] [  1%] Built target mongodb_store_generate_messages_lisp
Building CXX object rosplan/rosplan_planning_system/CMakeFiles/rosplan_planning_system.dir/src/PlanningSystem.cpp.o
[  1%] Built target move_base_msgs_generate_messages_cpp
[  1%] Built target actionlib_msgs_generate_messages_lisp
[  1%] Built target mongodb_store_msgs_generate_messages_lisp
[  1%] Built target move_base_msgs_generate_messages_py
[  1%] Built target actionlib_msgs_generate_messages_py
[  1%] Built target actionlib_generate_messages_cpp
[  1%] Built target actionlib_generate_messages_py
[  1%] Built target move_base_msgs_generate_messages_lisp
[  1%] Built target mongodb_store_generate_messages_cpp
[  1%] Built target mongodb_store_generate_messages_py
[  1%] Built target mongodb_store_msgs_generate_messages_py
[  1%] Built target actionlib_msgs_generate_messages_cpp
[  1%] Built target actionlib_generate_messages_lisp
[  1%] Built target _occupancy_grid_utils_generate_messages_check_deps_OverlayClouds
[  1%] Built target nav_msgs_generate_messages_cpp
[  1%] Built target sensor_msgs_generate_messages_cpp
[  1%] Built target _occupancy_grid_utils_generate_messages_check_deps_LocalizedCloud
[  1%] Built target _occupancy_grid_utils_generate_messages_check_deps_NavigationFunction
[  1%] Built target nav_msgs_generate_messages_lisp
[  1%] Built target sensor_msgs_generate_messages_lisp
[  1%] Built target nav_msgs_generate_messages_py
[  1%] Built target sensor_msgs_generate_messages_py
[  1%] Building CXX object rosplan/rosplan_planning_system/CMakeFiles/rosplan_planning_system.dir/src/SimplePlanDispatcher.cpp.o
In file included from /home/viki/ROSPlan/src/rosplan/rosplan_planning_system/src/PlanningSystem.cpp:1:0:
/home/viki/ROSPlan/src/rosplan/rosplan_planning_system/include/rosplan_planning_system/PlanningSystem.h:6:50: fatal error: rosplan_dispatch_msgs/ActionDispatch.h: No such file or directory
 #include "rosplan_dispatch_msgs/ActionDispatch.h"
                                                  ^
compilation terminated.
In file included from /home/viki/ROSPlan/src/rosplan/rosplan_planning_system/include/rosplan_planning_system/SimplePlanDispatcher.h:1:0,
                 from /home/viki/ROSPlan/src/rosplan/rosplan_planning_system/src/SimplePlanDispatcher.cpp:1:
/home/viki/ROSPlan/src/rosplan/rosplan_planning_system/include/rosplan_planning_system/PlanDispatcher.h:5:50: fatal error: rosplan_dispatch_msgs/ActionDispatch.h: No such file or directory
 #include "rosplan_dispatch_msgs/ActionDispatch.h"
                                                  ^
compilation terminated.
make[2]: *** [rosplan/rosplan_planning_system/CMakeFiles/rosplan_planning_system.dir/src/PlanningSystem.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[  1%] Built target visualization_msgs_generate_messages_lisp
[  1%] Built target tf_generate_messages_py
[  1%] Built target tf_generate_messages_cpp
[  1%] Built target tf_generate_messages_lisp
[  1%] Built target tf2_msgs_generate_messages_cpp
[  1%] Built target tf2_msgs_generate_messages_lisp
[  1%] Built target tf2_msgs_generate_messages_py
[  1%] Built target visualization_msgs_generate_messages_cpp
[  1%] Built target visualization_msgs_generate_messages_py
[  1%] Built target kobuki_msgs_generate_messages_lisp
[  1%] Built target kobuki_msgs_generate_messages_cpp
[  1%] Built target kobuki_msgs_generate_messages_py
[  2%] Generating Lisp code from rosplan_dispatch_msgs/CompletePlan.msg
[  2%] Generating Lisp code from rosplan_dispatch_msgs/ActionFeedback.msg
[  3%] Generating Lisp code from rosplan_dispatch_msgs/ActionDispatch.msg
[  3%] Built target rosplan_dispatch_msgs_generate_messages_lisp
make[2]: *** [rosplan/rosplan_planning_system/CMakeFiles/rosplan_planning_system.dir/src/SimplePlanDispatcher.cpp.o] Error 1
make[1]: *** [rosplan/rosplan_planning_system/CMakeFiles/rosplan_planning_system.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  3%] [  4%] Generating C++ code from rosplan_dispatch_msgs/CompletePlan.msg
Generating C++ code from rosplan_dispatch_msgs/ActionFeedback.msg
[  4%] Generating C++ code from rosplan_dispatch_msgs/ActionDispatch.msg
[  4%] Built target rosplan_dispatch_msgs_generate_messages_cpp
make: *** [all] Error 2
Invoking "make" failed

m312z, 2016-04-10:
It looks like catkin_make is trying to build the planning system before the messages are built (upon which it depends.) This usually happened when multiple cores used to build -- although that shouldn't be the case any more.

Try building with only one core:
`catkin_make -j1`

MahmoudElShazly, 2016-04-10:
catkin_make -j1
worked perfectly.

Should not we adjust the dependencies to allow compilation with more than one thread?

m312z, 2016-04-10:
Hello,

The dependencies are already stated. This seems to be a problem with running catkin_make with more than one thread.

rosplan_planning_system has a dependency on rosplan_dispatch_msgs in package.xml and in CMakelists. (Perhaps make sure you have the latest version of the repo? Just in case.)

You can also use the tool "catkin build". This correctly builds with multiple threads.
You can get it by installing:
`sudo apt-get install python-catkin-tools`
Delete the build and devel folder and build using this instead. You can also use this tool to easily build individual packages, eg. `catkin build rosplan_planning_system`

Michael

MahmoudElShazly, 2016-04-11:
Thanks, catkin build worked also.

closed 2016-04-11 by MahmoudElShazly