KCL-Planning/ROSPlan #25: turtlebot_explore.bash warnings

MahmoudElShazly, 2016-04-12:
I have successfully installed rosplan and ran:

roslaunch rosplan_planning_system planning_system_knowledge.launch 

after that i ran:
sh /src/rosplan/rosplan_demos/scripts/turtlebot_explore.bash

I got the generated plan file as:

Warnings encountered when parsing Domain/Problem File

Errors: 0, warnings: 10
/home/viki/ROSPlan/src/rosplan/rosplan_planning_system/common/problem.pddl: line: 9: Warning: Undeclared symbol: wp0
/home/viki/ROSPlan/src/rosplan/rosplan_planning_system/common/problem.pddl: line: 10: Warning: Undeclared symbol: wp1
/home/viki/ROSPlan/src/rosplan/rosplan_planning_system/common/problem.pddl: line: 11: Warning: Undeclared symbol: wp2
/home/viki/ROSPlan/src/rosplan/rosplan_planning_system/common/problem.pddl: line: 12: Warning: Undeclared symbol: wp3
/home/viki/ROSPlan/src/rosplan/rosplan_planning_system/common/problem.pddl: line: 13: Warning: Undeclared symbol: wp4
/home/viki/ROSPlan/src/rosplan/rosplan_planning_system/common/problem.pddl: line: 14: Warning: Undeclared symbol: wp5
/home/viki/ROSPlan/src/rosplan/rosplan_planning_system/common/problem.pddl: line: 15: Warning: Undeclared symbol: wp6
/home/viki/ROSPlan/src/rosplan/rosplan_planning_system/common/problem.pddl: line: 16: Warning: Undeclared symbol: wp7
/home/viki/ROSPlan/src/rosplan/rosplan_planning_system/common/problem.pddl: line: 17: Warning: Undeclared symbol: wp8
/home/viki/ROSPlan/src/rosplan/rosplan_planning_system/common/problem.pddl: line: 18: Warning: Undeclared symbol: wp9
Object with unknown type: wp0

what do you think the reason is?

MahmoudElShazly, 2016-04-12:
And this is the problem file:

(define (problem turtlebot_demo_task)
(:domain turtlebot_demo)
(:objects
    kenny - robot
)
(:init
)
(:goal (and
    (visited wp0)
    (visited wp1)
    (visited wp2)
    (visited wp3)
    (visited wp4)
    (visited wp5)
    (visited wp6)
    (visited wp7)
    (visited wp8)
    (visited wp9)
)))

m312z, 2016-05-23:
Hello,

Sorry, I didn't see this earlier.
```
roslaunch rosplan_planning_system planning_system_knowledge.launch
```
launches the Knowledge Base and Scene Database. However it does not include any waypoint generation. This means that no waypoints have been generated and added to the PDDL model.

The script is meant for use with the `rosplan_demos turtlebot.launch`. If you have a look at this launch file, and the included links within, you should find `rosplan_interface_mapping`. This is used to generate waypoints.

The script:
- calls the mapping node to generate waypoints, which adds them to the PDDL model in the Knowledge Base.
- adds the PDDL goals to the knowledge base (e.g. (visited wp9)).
- and calls the planner.

closed 2017-01-30 by m312z