KCL-Planning/ROSPlan #28: Suggestion - Minor modifications to improve the system.

dgerod, 2016-06-19:
I have been working with ROSPlan in my research and I like it a lot. Due to my work with it I have a couple of suggestions that in my opinion improve the system and are easy to do. 

1)  The _domain.pddl_ files of the demos should not be in _rosplan_planning_system/common_ but in _rosplan_demos/common_ as they are part of the demos and not of the system.

[image]

2)  I suggest to create a new package inside _rosplan_ meta-package named as _rosplan_config_, to store general configuration information and tools instead being distributed inside the different components, and example of that are _MongoDB_ and _popf_ planner binary. In addition, in case that a user would like to use a different DB or planner he only need to update this package (assuming that update code is not needed).  

[image]

I have already modified the system in my local repository as they are easy to do.

m312z, 2016-06-19:
Hello,

Both are good ideas.

For (2) there would also need to be some update of code. I am currently writing a tutorial to do these things. However, I still think it is a good idea to separate out other libraries and tools from the ROSPlan packages.

Are you able to make a [pull request](https://help.github.com/articles/using-pull-requests/) with the changes?

Michael

dgerod, 2016-06-19:
Yes, sure. 
Let me integrate my repository with your last updates before doing it.

closed 2016-06-22 by dgerod