KCL-Planning/ROSPlan #30: PDDL problem file is incorrectly generated

dgerod, 2016-06-22:
The problem file is not generated correctly, you could see the output below. The issue is that the arguments of the facts are written in the inverse order in the initial state. 
```
define (problem gripper_four_balls_task)
(:domain gripper_four_balls)
(:objects
    ball1 ball2 ball3 ball5 ball4 - ball
    left right - gripper
    room1 room2 - room
)
(:init
    (at-ball room1 ball1)
    (at-ball room1 ball3)
    (at-ball room2 ball2)
    (at-ball room2 ball4)
    (at-robby room1)
    (free left)
    (free right)
)
(:goal (and
    (at-ball ball1 room2)
    (at-ball ball2 room2)
    (at-ball ball3 room2)
    (at-ball ball4 room2)
)))
```
They should be written like in the goal: _(at-ball ball1 room2)_ but they are written: _(at-ball room1 ball1)_. However, I am passing the arguments in the same way when I setting facts and goals; I use "rosplan_interface" package to do that.

This happens after integrating last modifications of ROSPlan in my code. I have checked the source code and reverting _ [PDDLProblemGeneration.cpp](https://github.com/KCL-Planning/ROSPlan/blob/0ba92fcd0710a5433b48ebc70f4855bc68f323c6/rosplan_planning_system/src/PDDLProblemGenerator.cpp)_ to version 0ba92fcd0710a5433b48ebc70f4855bc68f323c6 the file is correctly generated; it seems an issue in finding the PDDL parameters in the Knowledge Items.

Please, could you check and confirm it? 
Thanks.

m312z, 2016-06-22:
I'm guessing this is the change with the problem:
https://github.com/KCL-Planning/ROSPlan/commit/0acfeeac9c1ad6bbd69e7046af5beb9c6dc1c20b

Before, the problem generation was iterating through the labels of the predicate from the domain file, finding the matching key in the knowledge item, and writing the value.

After, the problem generation ignores the domain file and simply writes the values in the knowledge item in the order they appear. In theory, this should not cause problems, as PDDL predicates define the order of their parameters, and the array of key-value  parameters in KnowledgeItem is also an ordered list.

However, this relies on users describing propositions with the same parameter order as the predicate. I've not used rosplan_interface, but my guess is that the dictionary in python is re-ordering them.

The best solution will be to go back to checking the parameter labels explicitly. (And fixing the code that necessitated this change in the first place.)

I'll start on this now.

Michael

dgerod, 2016-06-22:
OK, thanks.

m312z, 2016-06-22:
Hello,

I have pushed a fix. The problem should be fixed now.

Note that now you must have the correct parameter labels for a proposition to appear in the intial state. For example,
```
v -> my_robot
wp -> waypoint0
```
will work, but not:
```
robot -> my_robot
to -> waypoint0
```
This is important if you are updating the state due to the outcome of actions!
The sample action interfaces I've included in ROSPlan update the state automatically based on the operator description. However, the operator description does not use the same parameter labels as the predicate.
This is now fixed (properly). The action interface fetches and stores the predicate details using the service: `/kcl_rosplan/get_domain_predicate_details`

If you have any state updates of your own, you should check exactly what propositions you are trying to add to the PDDL model. The error would only appear when the system replans, tries to generate an initial state, and finds that the propositions do not have the correct labels.

Let me know if it works, and we can close this one.

Thanks,
Michael

dgerod, 2016-06-22:
Yes, it works! Thanks for fixing.

closed 2016-06-22 by dgerod