KCL-Planning/ROSPlan #32: What is mongodb used for?

cdondrup, 2016-09-13:
Hi,

As you might have seen from #31, I recently started using rosplan for our project and started having a look at some parts of the code. Regarding `mongodb` or `message_store` in particular, I discovered that it is started in all the launch files that contain the knowledge base, but apart from [this bit of code](https://github.com/KCL-Planning/ROSPlan/blob/5b32f44a5d397a370f18c67b80478fc3d52ff72a/rosplan_knowledge_base/src/KnowledgeBase.cpp#L484-L486):

```cpp
	// wait for and clear mongoDB 
	ROS_INFO("KCL: (KB) Waiting for MongoDB");
	ros::service::waitForService("/message_store/delete",-1);
	system("mongo message_store --eval \"printjson(db.message_store.remove())\"");
``` 

it is not used in the knowledge base itself. I know that it is used in the movebase and mapping interfaces, but not in the knowledge base, hence it wouldn't be necessary to start `mongodb` at all if the interfaces are not used. This way, you wouldn't have to kill the system mongodb before starting the planning system and it would reduce dependencies as well.

So my question is, why is it started every time and are there plans to use it? Because currently, the knowledge base is not persistent. Moreover, since the `message_store` is deleted on every start-up, not even the mapping and movebase interface are able store data persistently. So what is the reasoning behind using it and why is the `message_store` deleted when the knowledge base is started?

m312z, 2016-09-14:
Hello,

> So my question is, why is it started every time and are there plans to use it?

The Knowledge Base only stores the PDDL model. It does not store any additional information that might be required during the dispatch of actions. For example, the PDDL model might describe the space as abstract waypoints, and the goto_waypoint action must still be dispatched with real coordinates.

MongoDB is used to store these real values, or other information, that is outwith the PDDL model. In the rosplan_interface_movebase/mapping it is used to describe waypoints as above.
Key: PDDL instance name; value: Real world information.

It is true that if there is no need for such extra information, then the message store could be removed from the launch files and not used.

> why is the message_store deleted when the knowledge base is started?

The message store is not intended to be persistent, which is why it is cleared on startup. If this is not done, then the persistent data can cause some strange errors. For example, any PDDL instances which repeat an instance name used in a previous run on the same machine (for example, waypoint0) will be associated with erroneous data.

I hope this answers those questions.

Michael

cdondrup, 2016-09-14:
Hi Micheal,

Thank you very much for the answers.

I am just still a little confused because the knowledge base itself doesn't store or retrieve anything from the mongodb. Also, dropping the whole message store seems a bit drastic. I completely understand the reasoning behind not having waypoints persistent across environments but if I would store anything else in the message_store it would be deleted as well. So if I ever wanted to have a persistent knowledge base (to combat the fact that the knowledge base node might die and be respawened by roslaunch) than this would be impossible using the message_store. Maybe having a specific collection for the waypoints that can be dropped without any harm (or at least not as much) would be better?!

Also, from what I found is that mongodb is (at least for the code in this repository) only used to store the x,y,theta positions of the waypoints based on their name in the mapping component and to retrieve them in the movebase component. So all this seems rather specific to the example given. Of course using it this way is very convenient to retrieve the coordinates based on the abstract name but to run the planning system itself, running mongodb and deleting the message_store is technically not necessary if you don't want to use the provided interfaces, right?

I've been running the `planning_system_knowledgebase.launch` file successfully without any mongodb after commenting the line where it deletes the message_store. So apart from not being able to run the mapping and movebase interface, is there any other detriment to not using mongodb? Are there any plans to make the knowledgebase persistent at some point? If not, I'd be happy to look into this and of course share it if it is of interest.

Sorry for all the questions. Your help is very much appreciated.

cdondrup, 2016-09-14:
Just read your answer to my pull request. So if there is no harm in removing it, then I will use it without for now but I will still have a look at making the knowledge base persistent. In case you are interested in this, I will open a pull request as soon as it works.

Thank you for your time!

closed 2016-09-14 by cdondrup

m312z, 2016-09-14:
> Maybe having a specific collection for the waypoints that can be dropped without any harm (or at least not as much) would be better?!

This would be the best solution. The mapping component that is responsible for adding waypoint PDDL objects to the knowledge base and their corresponding coordinates to the message store would also be responsible for cleaning the old data. This component has no persistent storage of its own, so the trick will be in storing the PDDL type of each instance in the message store as well...

> ... So all this seems rather specific to the example given. Of course using it this way is very convenient to retrieve the coordinates based on the abstract name but to run the planning system [...]

Right. However, this type of use appears in most scenarios. For example, we store point clouds, stamped poses, and categories of objects based on their abstract name as well. In general, the abstract PDDL references real data that must be stored _somewhere_.

Don't worry about questions. You are certainly improving the framework by interrogating the ragged edges...

Michael