KCL-Planning/ROSPlan #41: Rosplan Demo (TurtleBot) - Error after generating waypoints

EsdrasChaves, 2017-03-16:
Hi there,

I'm trying to run the Demo with turtlebot, everything is working fine but when I run 
`sh src/rosplan/rosplan_demos/scripts/turtlebot_explore.bash` 
I get this output: 
```
waypoints: ['wp0', 'wp1', 'wp2', 'wp3', 'wp4']
success: True
success: True
success: True
success: True
success: True
success: True
success: True
success: True
ERROR: service [/kcl_rosplan/planning_server] responded with an error:
```

I can see the waypoints in RViz and run 
`rqt --standalone rosplan_rqt.dispatcher.ROSPlanDispatcher`
but then nothing happens.

I'm a beginner at ROS and I really enjoyed your work, I'm also researching Planning at the University xD

Cheers,
Esdras Chaves

DanniloMiranda, 2017-03-16:
I have the same problem.
The plan is not being generated. 
The problem file don't exist. 

The plan generated is: 
`` Number of literals: 2
Constructing lookup tables:
A problem has been encountered, and the problem has been deemed unsolvable
--------------------------------------------------------------------------
The goal fact:
(visited wp1)

...cannot be found either in the initial state, as an add effect of an
 action, or as a timed initial literal.  As such, the problem has been deemed
unsolvable.``

m312z, 2017-03-17:
The output you have included is that the planner can't find a solution for the problem. "(visited wp1)" would only appear in a problem file, and not the domain. This means that at least the problem is being generated somewhere. According to the launch file it should be found in:
- rosplan_planning_system/common/problem.pddl

Why the problem cannot be solved is another matter...

Can you find the output line that begins:
- "KCL: (PS) ([problem name]) Running: [command string]"
and try copy/paste to run the command string directly in the console? This should reveal (1) if the problem really exists, (2) where it is, (3) which domain file is used, and (4) perhaps more planner output.

DerekLong101, 2017-03-17:
Hi Michael

Actually, that output shows the domain exists, the problem exists and the planner was correctly called. The problem file must contain an unsolvable problem. This probably means that the locations are not connected together. It would be good to make sure that when the planner does not produce a plan, ROSPlan correctly enters a state in which the problem was considered unsolvable and there is no plan to execute.

Cheers

Derek

m312z, 2017-03-17:
Hello Derek!

It shows that _a_ problem and domain file exist, but they may not be the ones that we expect...

The "goto_waypoint" action does not have a connected-ness precondition, only:

> ;; Move between any two waypoints, avoiding terrain
> (:durative-action goto_waypoint
> ...
> 	:condition (and (at start (robot_at ?v ?from)))
> ...
> )

However, I just had a thought. There is another domain "domain_turtlebot" which has another precondition: that the turtlebot is undocked. This domain is used with the automatic docking feature of the turtlebot.

I bet that this domain is being used, instead of "domain_turtlebot_demo". This would mean that, since neither the "undocked" or "docked" (and even "localised") facts are in the initial state, the planner cannot find any applicable actions.

I've just changed the domain file parameter in "rosplan_planning_system/interface_planning_system.launch".
Can you check your local copies to see if that is the problem?

Michael

DanniloMiranda, 2017-03-20:
That was the problem. 
I changed the domain file parameter and now is working.

EsdrasChaves, 2017-03-20:
It worked xD
Thanks for your support!

Closing the topic.

closed 2017-03-20 by EsdrasChaves