KCL-Planning/ROSPlan #47: Compilation error

mokhtarivahid, 2017-05-02:
[Ubuntu 12.04 fully updated, ROS Hydro]

Hello, 
I am getting the following errors in installing ros_plan:

`vahid@vahid-ThinkPad-W520:~/ROSPlan$ catkin_make
Base path: /home/vahid/ROSPlan
Source space: /home/vahid/ROSPlan/src
Build space: /home/vahid/ROSPlan/build
Devel space: /home/vahid/ROSPlan/devel
Install space: /home/vahid/ROSPlan/install
####
#### Running command: "make cmake_check_build_system" in "/home/vahid/ROSPlan/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/vahid/ROSPlan/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/hydro
-- This workspace overlays: /opt/ros/hydro
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Python version: 2.7
-- Using Debian Python package layout
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/vahid/ROSPlan/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- catkin 0.5.90
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 11 packages in topological order:
-- ~~  - rosplan (metapackage)
-- ~~  - rosplan_demos
-- ~~  - rosplan_rqt
-- ~~  - rosplan_dispatch_msgs
-- ~~  - rosplan_knowledge_msgs
-- ~~  - rosplan_planning_system
-- ~~  - rosplan_interface_movebase
-- ~~  - occupancy_grid_utils
-- ~~  - rosplan_interface_mapping
-- ~~  - rosplan_interface_turtlebot2
-- ~~  - rosplan_knowledge_base
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'rosplan'
-- ==> add_subdirectory(rosplan/rosplan)
-- +++ processing catkin package: 'rosplan_demos'
-- ==> add_subdirectory(rosplan/rosplan_demos)
-- +++ processing catkin package: 'rosplan_rqt'
-- ==> add_subdirectory(rosplan/rosplan_rqt)
-- +++ processing catkin package: 'rosplan_dispatch_msgs'
-- ==> add_subdirectory(rosplan/rosplan_dispatch_msgs)
-- Using these message generators: gencpp;genlisp;genpy
-- Generating .msg files for action rosplan_dispatch_msgs/Plan /home/vahid/ROSPlan/src/rosplan/rosplan_dispatch_msgs/action/Plan.action
-- rosplan_dispatch_msgs: 10 messages, 1 services
-- +++ processing catkin package: 'rosplan_knowledge_msgs'
-- ==> add_subdirectory(rosplan/rosplan_knowledge_msgs)
-- Using these message generators: gencpp;genlisp;genpy
-- rosplan_knowledge_msgs: 5 messages, 14 services
-- +++ processing catkin package: 'rosplan_planning_system'
-- ==> add_subdirectory(rosplan/rosplan_planning_system)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'rosplan_interface_movebase'
-- ==> add_subdirectory(rosplan/rosplan_interface_movebase)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'occupancy_grid_utils'
-- ==> add_subdirectory(occupancy_grid_utils)
-- Using these message generators: gencpp;genlisp;genpy
-- occupancy_grid_utils: 3 messages, 0 services
-- +++ processing catkin package: 'rosplan_interface_mapping'
-- ==> add_subdirectory(rosplan/rosplan_interface_mapping)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'rosplan_interface_turtlebot2'
-- ==> add_subdirectory(rosplan_interface_turtlebot2)
[... 117 lines cut ...]
In file included from /home/vahid/ROSPlan/src/rosplan/rosplan_planning_system/src/PlanParsing/CLGPlanParser.cpp:1:0:
/home/vahid/ROSPlan/src/rosplan/rosplan_planning_system/include/rosplan_planning_system/CLGPlanParser.h:2:30: fatal error: std_srvs/Trigger.h: No such file or directory
compilation terminated.
make[2]: *** [rosplan/rosplan_planning_system/CMakeFiles/planner.dir/src/PlanParsing/CFFPlanParser.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 71%] Building CXX object rosplan/rosplan_planning_system/CMakeFiles/rosplan_planning_system.dir/src/PlanParsing/POPFEsterelPlanParser.cpp.o
make[2]: *** [rosplan/rosplan_planning_system/CMakeFiles/rosplan_planning_system.dir/src/PlanParsing/CFFPlanParser.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [rosplan/rosplan_planning_system/CMakeFiles/planner.dir/src/PlanParsing/CLGPlanParser.cpp.o] Error 1
make[2]: *** [rosplan/rosplan_planning_system/CMakeFiles/rosplan_planning_system.dir/src/PlanParsing/CLGPlanParser.cpp.o] Error 1
make[2]: *** [rosplan/rosplan_planning_system/CMakeFiles/planner.dir/src/PlanDispatch/EsterelPlanDispatcher.cpp.o] Error 1
make[2]: *** [rosplan/rosplan_planning_system/CMakeFiles/rosplan_planning_system.dir/src/PlanDispatch/EsterelPlanDispatcher.cpp.o] Error 1
make[2]: *** [rosplan/rosplan_planning_system/CMakeFiles/planner.dir/src/PlanningSystem.cpp.o] Error 1
make[1]: *** [rosplan/rosplan_planning_system/CMakeFiles/planner.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [rosplan/rosplan_planning_system/CMakeFiles/rosplan_planning_system.dir/src/PlanningSystem.cpp.o] Error 1
make[1]: *** [rosplan/rosplan_planning_system/CMakeFiles/rosplan_planning_system.dir/all] Error 2
make: *** [all] Error 2
Invoking "make" failed
vahid@vahid-ThinkPad-W520:~/ROSPlan$ 
`

Thank you for any help.
Best,
Vahid

m312z, 2017-05-02:
Looks like there is a missing "std_srvs/Trigger.h" which was added in ROS Indigo.
(http://wiki.ros.org/std_srvs)

If upgrading to Indigo is not an option, it should be sufficient to replace this service with "std_srvs/Empty.h", which is the only std_srv in Hydro.

mokhtarivahid, 2017-05-02:
I just replaced all "std_srvs/Trigger.h" by "std_srvs/Empty.h", and it worked perfectly.
I close the issue.
Thank you.

closed 2017-05-02 by mokhtarivahid