KCL-Planning/ROSPlan #48: cannot launch node of type [rosplan_knowledge_base/knowledgeBase] ...

mokhtarivahid, 2017-05-02:
Hello,

I am getting the following errors (errors are in bold):

`vahid@vahid-ThinkPad-W520:~$ roslaunch rosplan_demos turtlebot.launch 
... logging to /home/vahid/.ros/log/1f412ed4-2f39-11e7-9670-0024d7ce5a44/roslaunch-vahid-ThinkPad-W520-15251.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://vahid-ThinkPad-W520:43553/

SUMMARY
========

PARAMETERS
 * /amcl/base_frame_id
 * /amcl/gui_publish_rate
 * /amcl/initial_pose_a
 * /amcl/initial_pose_x
 * /amcl/initial_pose_y
 * /amcl/kld_err
 * /amcl/kld_z
 * /amcl/laser_lambda_short
 * /amcl/laser_likelihood_max_dist
 * /amcl/laser_max_beams
 * /amcl/laser_max_range
 * /amcl/laser_model_type
 * /amcl/laser_sigma_hit
 * /amcl/laser_z_hit
 * /amcl/laser_z_max
 * /amcl/laser_z_rand
 * /amcl/laser_z_short
 * /amcl/max_particles
 * /amcl/min_particles
 * /amcl/odom_alpha1
 * /amcl/odom_alpha2
 * /amcl/odom_alpha3
 * /amcl/odom_alpha4
 * /amcl/odom_alpha5
 * /amcl/odom_frame_id
 * /amcl/odom_model_type
 * /amcl/recovery_alpha_fast
 * /amcl/recovery_alpha_slow
 * /amcl/resample_interval
 * /amcl/transform_tolerance
 * /amcl/update_min_a
 * /amcl/update_min_d
 * /amcl/use_map_topic
 * /bumper2pointcloud/pointcloud_radius
 * /cmd_vel_mux/yaml_cfg_file
 * /depthimage_to_laserscan/output_frame_id
 * /depthimage_to_laserscan/range_min
 * /depthimage_to_laserscan/scan_height
 * /move_base/TrajectoryPlannerROS/acc_lim_theta
 * /move_base/TrajectoryPlannerROS/acc_lim_x
 * /move_base/TrajectoryPlannerROS/acc_lim_y
 * /move_base/TrajectoryPlannerROS/dwa
 * /move_base/TrajectoryPlannerROS/gdist_scale
 * /move_base/TrajectoryPlannerROS/heading_lookahead
 * /move_base/TrajectoryPlannerROS/holonomic_robot
 * /move_base/TrajectoryPlannerROS/max_vel_theta
 * /move_base/TrajectoryPlannerROS/max_vel_x
 * /move_base/TrajectoryPlannerROS/max_vel_y
 * /move_base/TrajectoryPlannerROS/meter_scoring
 * /move_base/TrajectoryPlannerROS/min_in_place_vel_theta
 * /move_base/TrajectoryPlannerROS/min_vel_theta
 * /move_base/TrajectoryPlannerROS/min_vel_x
 * /move_base/TrajectoryPlannerROS/min_vel_y
 * /move_base/TrajectoryPlannerROS/occdist_scale
[... 266 lines cut ...]
Error [Visual.cc:2072] No mesh specified
Error [ModelDatabase.cc:408] Unable to download model[model://cube_20k/meshes/cube_20k.stl]
Error [SystemPaths.cc:371] File or path does not exist[""]
Error [Visual.cc:2072] No mesh specified**
[ INFO] [1493730877.318801892, 1.590000000]: Loading from pre-hydro parameter style
[ INFO] [1493730877.360633013, 1.630000000]: Using plugin "static_layer"
[ INFO] [1493730877.428544205, 1.700000000]: Requesting the map...
[ INFO] [1493730877.626382263, 1.900000000]: Resizing costmap to 231 X 194 at 0.050000 m/pix
[ INFO] [1493730877.725980765, 2.000000000]: Received a 231 X 194 map at 0.050000 m/pix
[ INFO] [1493730877.730298825, 2.000000000]: Using plugin "obstacle_layer"
[ INFO] [1493730877.732612955, 2.000000000]:     Subscribed to Topics: scan bump
[ INFO] [1493730877.800121512, 2.070000000]: Using plugin "inflation_layer"
[ INFO] [1493730877.954134676, 2.220000000]: Loading from pre-hydro parameter style
[ INFO] [1493730877.997926242, 2.270000000]: Using plugin "obstacle_layer"
[ INFO] [1493730878.060802801, 2.330000000]:     Subscribed to Topics: scan bump
[ INFO] [1493730878.121963235, 2.390000000]: Using plugin "inflation_layer"
[ INFO] [1493730878.237420707, 2.500000000]: Created local_planner base_local_planner/TrajectoryPlannerROS
[ INFO] [1493730878.260525950, 2.530000000]: Sim period is set to 0.20
[ INFO] [1493730879.034581690, 3.300000000]: Recovery behavior will clear layer obstacles
[ INFO] [1493730879.103149710, 3.370000000]: Recovery behavior will clear layer obstacles
[ INFO] [1493730879.145982205, 3.410000000]: odom received!
`

Thank you for your help.
Best

m312z, 2017-05-02:
The errors of type:
> ERROR: cannot launch node of type [rosplan_knowledge_base/knowledgeBase]: can't locate node [knowledgeBase] in package [rosplan_knowledge_base]
will be fixed by sourcing the workspace before launching. If not, make sure that the packages are properly compiled (look for them in the devel folder) and try to find them using roscd.

The other errors are from Gazebo, which might be having problems downloading models. You can download them manually and put them in the right directory. I found the same issue here:
http://answers.gazebosim.org/question/4399/turtlebot_gazebo-world-not-loading-models/

Hope this helps,
Michael

mokhtarivahid, 2017-05-02:
Hi Michael,
It was resolved by sourcing the workspace.
I also downloaded the models and added to gazebo, and now, seems it works well.

I am facing a new error in running "rqt --standalone rosplan_rqt.dispatcher.ROSPlanDispatcher" as follow:

`vahid@vahid-ThinkPad-W520:~$ rqt --standalone rosplan_rqt.dispatcher.ROSPlanDispatcher
RosPluginProvider.load(rosplan_rqt/ROSPlanDispatcher) exception raised in __builtin__.__import__(rosplan_rqt.dispatcher, [ROSPlanDispatcher]):
Traceback (most recent call last):
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 77, in load
    module = __builtin__.__import__(attributes['module_name'], fromlist=[attributes['class_from_class_type']], level=0)
  File "/home/vahid/ROSPlan/src/rosplan/rosplan_rqt/src/rosplan_rqt/dispatcher.py", line 3, in <module>
    from .ROSPlanDispatcher import PlanViewWidget
  File "/home/vahid/ROSPlan/src/rosplan/rosplan_rqt/src/rosplan_rqt/ROSPlanDispatcher.py", line 13, in <module>
    from rosplan_dispatch_msgs.msg import *
ImportError: No module named rosplan_dispatch_msgs.msg

PluginManager._load_plugin() could not load plugin "rosplan_rqt/ROSPlanDispatcher":
Traceback (most recent call last):
  File "/opt/ros/hydro/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 98, in load
    self._load()
  File "/opt/ros/hydro/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 54, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 83, in load
    raise e
ImportError: No module named rosplan_dispatch_msgs.msg
`

Do you have any idea?
Thank you.
Vahid

mokhtarivahid, 2017-05-02:
I forgot to source it again.
It is Ok now.
Thank you.

closed 2017-05-02 by mokhtarivahid

dgerod, 2021-10-30:
It means to use "source" in your local workspace instead to the ROS distribution one. So, calling "source devel/setup.bash" instead "source /opt/ros/DISTRIBUTION/setup.bash"

This command add all your local package to the ROS path so ROS tools is able to find them. To check if your local packages are correctly added you could use "roscd NAME_OF_YOUR_PACKAGE" or "rospack list | grep NAME_OF_YOUR_PACKAGE."