KCL-Planning/ROSPlan #49: Error while building

AliSiddiqui4, 2017-05-12:
Error when catkin_make the ROSPlan folder 

```
Could not find a package configuration file provided by "move_base_msgs"
  with any of the following names:

    move_base_msgsConfig.cmake
    move_base_msgs-config.cmake

  Add the installation prefix of "move_base_msgs" to CMAKE_PREFIX_PATH or set
  "move_base_msgs_DIR" to a directory containing one of the above files.  If
  "move_base_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  rosplan/rosplan_interface_movebase/CMakeLists.txt:5 (find_package)

-- Could not find the required component 'move_base_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "move_base_msgs"
  with any of the following names:

    move_base_msgsConfig.cmake
    move_base_msgs-config.cmake

  Add the installation prefix of "move_base_msgs" to CMAKE_PREFIX_PATH or set
  "move_base_msgs_DIR" to a directory containing one of the above files.  If
  "move_base_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  rosplan/rosplan_interface_movebase/CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/muhammad/ROSPlan/build/CMakeFiles/CMakeOutput.log".
See also "/home/muhammad/ROSPlan/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

```

m312z, 2017-05-12:
Hello,

The error is right at the top: could not find package "move_base_msgs". This package is included in the desktop-full installation of ROS.

Alternatively, you can install it separately using:
sudo apt-get install ros-indigo-move-base

You might find there are some more missing packages if you haven't yet installed:
sudo apt-get install ros-indigo-desktop
or
sudo apt-get install ros-indigo-desktop-full

Hope this helps,
Michael

closed 2017-05-12 by m312z

AliSiddiqui4, 2017-05-12:
ok but now I am getting this error even though I have hector installed: 

`Could not find a package configuration file provided by "hector_move_base"
`