KCL-Planning/ROSPlan #52: Problem Compilation ROSPlan (Invoking "make -j8 -l8" failed)

ismaisma, 2017-05-31:
Hi, 
I tried to re-install ROSPLAN but I have problem with catkin_make, I have the following message: 

`Base path: /home/srp/ROSPlan
Source space: /home/srp/ROSPlan/src
Build space: /home/srp/ROSPlan/build
Devel space: /home/srp/ROSPlan/devel
Install space: /home/srp/ROSPlan/install
####
#### Running command: "make cmake_check_build_system" in "/home/srp/ROSPlan/build"
####
####
#### Running command: "make -j8 -l8" in "/home/srp/ROSPlan/build"
####
[  0%] [  0%] Built target diagnostic_msgs_generate_messages_py
Built target actionlib_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target _rosplan_dispatch_msgs_generate_messages_check_deps_PlanGoal
[  0%] [  0%] [  0%] Built target _rosplan_dispatch_msgs_generate_messages_check_deps_CompletePlan
Built target _rosplan_dispatch_msgs_generate_messages_check_deps_PlanActionFeedback
Built target _rosplan_dispatch_msgs_generate_messages_check_deps_PlanActionResult
[  0%] Built target _rosplan_dispatch_msgs_generate_messages_check_deps_PlanFeedback
[  0%] [  0%] Built target _rosplan_dispatch_msgs_generate_messages_check_deps_ActionFeedback
Built target actionlib_msgs_generate_messages_cpp
[  0%] [  0%] Built target _rosplan_dispatch_msgs_generate_messages_check_deps_PlanResult
Built target diagnostic_msgs_generate_messages_cpp
[  0%] Built target _rosplan_dispatch_msgs_generate_messages_check_deps_PlanningService
[  0%] [  0%] [  0%] Built target std_msgs_generate_messages_cpp
Built target diagnostic_msgs_generate_messages_lisp
Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target _rosplan_dispatch_msgs_generate_messages_check_deps_PlanActionGoal
[  0%] [  0%] Built target _rosplan_dispatch_msgs_generate_messages_check_deps_ActionDispatch
Built target _rosplan_dispatch_msgs_generate_messages_check_deps_PlanAction
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_GetDomainOperatorService
[  0%] [  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_RemoveWaypoint
Built target _rosplan_knowledge_msgs_generate_messages_check_deps_KnowledgeItem
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_Filter
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_KnowledgeUpdateServiceArray
[  0%] [  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_AddWaypoint
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_GenerateProblemService
Built target geometry_msgs_generate_messages_py
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_GetAttributeService
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_KnowledgeUpdateService
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_DomainFormula
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_GetDomainTypeService
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_Notification
[  0%] [  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_GetDomainAttributeService
Built target _rosplan_knowledge_msgs_generate_messages_check_deps_GetInstanceService
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_DomainOperator
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] [  0%] Built target geometry_msgs_generate_messages_lisp
Built target actionlib_generate_messages_lisp
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_KnowledgeQueryService
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] [  0%] Built target roscpp_generate_messages_py
Built target roscpp_generate_messages_lisp
[  0%] [  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_CreatePRM
Built target actionlib_generate_messages_py
[  0%] Built target _rosplan_knowledge_msgs_generate_messages_check_deps_GetDomainPredicateDetailsService
[  0%] [  0%] [  0%] [  0%] Built target actionlib_generate_messages_cpp
Built target std_srvs_generate_messages_cpp
Built target std_srvs_generate_messages_py
Built target _rosplan_knowledge_msgs_generate_messages_check_deps_GetDomainOperatorDetailsService
[  0%] [  0%] Built target std_srvs_generate_messages_lisp
Built target mongodb_store_generate_messages_lisp
[  0%] Built target move_base_msgs_generate_messages_py
[... 59 lines cut ...]
/home/srp/ROSPlan/src/rosplan/rosplan_planning_system/src/print_plan.cpp:47:24: error: range-based ‘for’ loops are not allowed in C++98 mode
   for (const auto &p : a.parameters) {
                        ^
/home/srp/ROSPlan/src/rosplan/rosplan_planning_system/src/print_plan.cpp:47:26: error: request for member ‘parameters’ in ‘a’, which is of non-class type ‘const int’
   for (const auto &p : a.parameters) {
                          ^
/home/srp/ROSPlan/src/rosplan/rosplan_planning_system/src/print_plan.cpp:48:17: error: request for member ‘value’ in ‘p’, which is of non-class type ‘const int’
    s += " " + p.value;
                 ^
/home/srp/ROSPlan/src/rosplan/rosplan_planning_system/src/print_plan.cpp:50:40: error: request for member ‘dispatch_time’ in ‘a’, which is of non-class type ‘const int’
   printf("%7.3f: (%s %s)  [%.3f]\n", a.dispatch_time, a.name.c_str(), s.c_str(), a.duration);
                                        ^
/home/srp/ROSPlan/src/rosplan/rosplan_planning_system/src/print_plan.cpp:50:57: error: request for member ‘name’ in ‘a’, which is of non-class type ‘const int’
   printf("%7.3f: (%s %s)  [%.3f]\n", a.dispatch_time, a.name.c_str(), s.c_str(), a.duration);
                                                         ^
/home/srp/ROSPlan/src/rosplan/rosplan_planning_system/src/print_plan.cpp:50:84: error: request for member ‘duration’ in ‘a’, which is of non-class type ‘const int’
   printf("%7.3f: (%s %s)  [%.3f]\n", a.dispatch_time, a.name.c_str(), s.c_str(), a.duration);
                                                                                    ^
make[2]: *** [rosplan/rosplan_planning_system/CMakeFiles/print_plan.dir/src/print_plan.cpp.o] Error 1
make[1]: *** [rosplan/rosplan_planning_system/CMakeFiles/print_plan.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`
I haven't understood really what is the problem 
Thank you

m312z, 2017-06-02:
The first error comes from here:

> /home/srp/ROSPlan/src/rosplan/rosplan_planning_system/src/print_plan.cpp:45:19: error: ISO C++ forbids declaration of ‘a’ with no type [-fpermissive]

I've updated the loop in print_plan to use a simple FOR loop which should be supported. It should now compile.

Michael

closed 2017-06-02 by m312z

ismaisma, 2017-06-04:
Hi, 
problem solved 
it works perfectly
Thank you  
Isma