KCL-Planning/ROSPlan #53: Weird behavior in KnowledgeComparitor::containsKnowledge

alessio-cpt, 2017-06-24:
Hi, I was working on ROSPlan a bit and I noticed something in ```KnowledgeComparitor::containsKnowledge```.

```c++
for(size_t i=0;i<a.values.size();i++) {
        // don't care about this parameter
	if("" == a.values[i].value) continue;

	// find matching object in parameters of b
	for(size_t i=0;i<b.values.size();i++) {
		if(! boost::iequals(a.values[i].key, b.values[i].key) ||
		   ! boost::iequals(a.values[i].value, b.values[i].value))
		{
			return false;
		}
	}
}
```
The index of the inner loop shadows the one of the outer, making the comparison inconsistent. 

I was having some errors while removing facts due to this, but since I heavily modified the knowledge base, I am not sure it is really an issue. Maybe I just messed it and cannot get the logic.

I fixed the function like this:

```c++
bool KnowledgeComparitor::containsKnowledge(const rosplan_knowledge_msgs::KnowledgeItem &a, const rosplan_knowledge_msgs::KnowledgeItem &b) {

        int matches = 0;

	if(a.knowledge_type != b.knowledge_type) return false;
	
	if(a.knowledge_type == rosplan_knowledge_msgs::KnowledgeItem::INSTANCE) {
			
	        // check instance knowledge
	        if(0!=a.instance_type.compare(b.instance_type)) return false;
		if(a.instance_name!="" && ! boost::iequals(a.instance_name, b.instance_name)) return false;

		} else {

		// check fact or function
		        if(a.attribute_name!="" && ! boost::iequals(a.attribute_name, b.attribute_name)) return false;
			if(a.is_negative != b.is_negative) return false;
			if(a.values.size() != b.values.size()) return false;

			for(size_t i=0;i<a.values.size();i++) {

				// don't care about this parameter
				if("" == a.values[i].value) {
                                        ++matches;
                                        continue;
                                }

				// find matching object in parameters of b
				for(size_t j=0;j<b.values.size();j++) {
				       if( boost::iequals(a.values[i].key, b.values[j].key) && 
                                           boost::iequals(a.values[i].value, b.values[j].value)) {
                                                ++matches;
					        break;
					}
				}
			}
		}
		return (matches == a.values.size());
	}
```

m312z, 2017-06-25:
Hello Alessio,

That does look wrong!

I remember that there are nested loops in case the parameters are not written in PDDL order. Your fix looks correct.

However, I've been using the queryKnowledge service and containsKnowledge method yesterday, and the behaviour was correct. This might mean there is another error somewhere correcting this one...

Oh hey! I see it: the inner loop is also using "int i" meaning that it is simply checking each pair of parameters multiple times.

Michael

m312z, 2017-06-25:
I just tested your code and put it into the main branch. Thanks!

Let me know if you find any other fixes, (or make other interesting changes).

Michael

closed 2017-06-25 by m312z

alessio-cpt, 2017-06-25:
Hi Micheal,

exactly, I wasn't sure because I also used queryKnowledge and containsKnowledge
 before and it always worked. I noticed it after I merged in my fork your
latest commits and updated from Indigo to Kinetic.

Of course I'll keep you updated!

Bye,
Alessio