KCL-Planning/ROSPlan #54: Problem with installation

bellerifabio, 2017-07-06:
Hello,
I'm trying to install ROSPlan on Ubuntu 14.04 using ROS indigo (full desktop version).

I've followed the installation guide and downloaded the source. When I run catkin_make (or catkin_make -j1), the process fails with the following error:

```
CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "move_base_msgs"
  with any of the following names:

    move_base_msgsConfig.cmake
    move_base_msgs-config.cmake

  Add the installation prefix of "move_base_msgs" to CMAKE_PREFIX_PATH or set
  "move_base_msgs_DIR" to a directory containing one of the above files.  If
  "move_base_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  rosplan/rosplan_interface_movebase/CMakeLists.txt:5 (find_package)

-- Could not find the required component 'move_base_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "move_base_msgs"
  with any of the following names:

    move_base_msgsConfig.cmake
    move_base_msgs-config.cmake

  Add the installation prefix of "move_base_msgs" to CMAKE_PREFIX_PATH or set
  "move_base_msgs_DIR" to a directory containing one of the above files.  If
  "move_base_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  rosplan/rosplan_interface_movebase/CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/fabio/catkin_ws/ROSPlan/build/CMakeFiles/CMakeOutput.log".
See also "/home/fabio/catkin_ws/ROSPlan/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

Thanks,
Fabio

alessio-cpt, 2017-07-06:
Hi,

it looks like `move_base_msgs` package is required but not installed on your system. 

Try installing it :
```sh
sudo apt-get install ros-indigo-move-base-msgs
```

bellerifabio, 2017-07-06:
Hi,

it works now and it has successfully compiled.

Thanks!

closed 2017-07-06 by bellerifabio