KCL-Planning/ROSPlan #55: ROSPlan for Lunar

swarupmohalik, 2017-08-23:
I am trying to install ROSPlan for Lunar.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.10
Release:	16.10
Codename:	yakkety

I understood that mongodb_store is replaced by warehouse_ros in Lunar. (Am I correct?)
Is there a port of ROSPlan to work with warehouse_ros, or only configurations will suffice?
In the latter case, where all I need to reconfigure?

Thanks in advance...

m312z, 2017-12-19:
Hello,

> I understood that mongodb_store is replaced by warehouse_ros in Lunar. (Am I correct?)

I did not know this.

I think an upgrade in this respect would be minor. The database is used to store messages that are not a part of the planning model. In the ROSPlan repo I think this only occurs in the mapping and movebase packages for waypoint coordinates.
- There the interface for MongoDB (message_store) will need to be updated to the warehouse_ros interface.
- The launch file will need updated to launch a warehouse_ros node instead of the old mongodb node.

Michael

oscar-lima, 2019-05-03:
see related issue #179 

and note the following:
- ROSPlan core implementation is no longer dependent on mongoDB
- rosplan_demos are now moved to a separate repo: https://github.com/KCL-Planning/rosplan_demos
- there are future short term plans to remove mongoDB from the demos in favor of ROS parameter server

closed 2019-05-03 by oscar-lima