KCL-Planning/ROSPlan #57: Problem with turtlebot example

bellerifabio, 2017-09-07:
Hello, I'm trying to run this example (https://github.com/KCL-Planning/ROSPlan/wiki/b.-Get-Started-with-ROSPlan), but when I run the command `sh src/rosplan/rosplan_demos/scripts/turtlebot_explore.bash` I get this error output:
`sh src/rosplan/rosplan_demos/scripts/turtlebot_explore.bash
waypoints: []
success: True
success: True
success: True
ERROR: service [/kcl_rosplan/planning_server] responded with an error: 
`

m312z, 2017-12-19:
Hello,

The first line:
"waypoints: []"
does not look right. It appears that no waypoints are generated, which will lead to an empty problem file and a goal referencing a non-existant object. What happens when instead of the explore script you run its first line?

rosservice call /kcl_rosplan/roadmap_server/create_prm "{nr_waypoints: 10, min_distance: 0.3, casting_distance: 2.0, connecting_distance: 8.0, occupancy_threshold: 10, total_attempts: 10}";

(Try to run it several times)

Michael

oscar-lima, 2019-05-03:
rosplan_demos are now moved to: https://github.com/KCL-Planning/rosplan_demos
if the issue still persists, please migrate it over there. Thanks

closed 2019-05-03 by oscar-lima