KCL-Planning/ROSPlan #59: PDDL parsing

bellerifabio, 2017-09-25:
I launched the planning system and the knowledge base nodes, providing my PDDL domain and problem. The system says:

`The supplied domain/problem file appear to violate part of the PDDL language specification. Specifically:

The planner will continue, but you may wish to fix your files accordingly.`

I don't understand what's the problem because the specification of the problem is empty; furthermore I used the PDDL files with another planner and it worked.
Can you help me please?

dgerod, 2017-11-01:
We need more information to help you:
- Which planner are you using? Are you using POPF as it is default planner of ROSPlan? 
- What domain and problem files are you using?

m312z, 2017-12-19:
Hello,

It could also be that you are intending to use an existing PDDL problem instance, but ROSPlan is generating a new (empty) problem from the knowledge base. If that is the case then you will need to check the launch parameters here:
http://kcl-planning.github.io/ROSPlan/documentation/planning_system_launch.html#pa

In particular "generate_default_problem" should be false. Its function is described in the link.

Michael