KCL-Planning/ROSPlan #60: [Q&A] POPF does not work correctly

dgerod, 2017-10-13:
I tried to solve some PDDL problems using POPF but it not works well for all plans, sometimes it said that plan is solved but does not return the actions and other times it frozen doing some kind of calculations when the solution is so simple. I make some simple test that produce these situations, you could find them in [this repository](https://github.com/dgerod/task-planning/tree/master/tests).

I know that POPF is an external tool so I should contact the responsible of it. However, I would like to ask here if you use a some tricks or policies to avoid this situations, as I suppose that this annoying situations are not happening to you.

Finally, I found a planner in Internet that solves my PDDL files without any problem, it is the [Fast Forward](https://github.com/dgerod/task-planning). As I would like to create the plan parser for ROSPlan, could you briefly explain me how to do it? I have already checked the PlanParser classes  that are provided with ROSPlan.

Thanks.

DerekLong101, 2017-10-13:
Hi Diego

If you run POPF on the test problems (I just checked the 4 in the popf tests directory) you will find that it solves the first and that the rest of them use negative preconditions (which POPF does not support). In fact, the negative preconditions are unnecessary and should be removed (and negative facts should not appear in the initial state - PDDL uses a closed world assumption). Also, in the action that has no precondition, you still have to put in the field (just empty). Problem 2 needs a robot to be solvable.

I have trivially modified and solved all 4 problems.

Just for information - POPF is the work of Amanda and Andrew Coles, Maria Fox and Derek Long.

Cheers

Derek

dgerod, 2017-10-15:
Thanks a lot for your answer!!! I tried to understand what was the problem in my files by searching in Internet but I did not find it. And due to that I was quite blocked in my work, so you help me a lot :-).

I have update all my PDDL test according to you feedback and, of course, they work now. And I have also updated the README in my repository to list all the authors of the POPF planner, sorry for forgetting to list you and other authors there.

closed 2017-10-15 by dgerod