KCL-Planning/ROSPlan #61: [Q&A] Create a PlanParser for a new planner

dgerod, 2017-10-18:
I would like to add the Fast Forward (FF) planner to ROSPlan so I have to created a plan parser. Could you briefly explain me how to do it? 

I have already checked PlanParser classes that are provided with ROSPlan, but I am not sure how I have to create the parser as there are different types in this directory, and I don't know why they are using different approaches. I have created the FF parser using POPFPlanParser as reference, do you think that this is correct? You could review my code [here](https://github.com/dgerod/task-planning/tree/master/rosplan).

Thanks.

alessio-cpt, 2017-10-18:
I did the same for Probe planner and it works.  `POPFPlanParser` and `PlanParser` are good references. You can avoid `POPFEsterelPlanParser` as long as you don't need Esterel plans. Concerning your code, I had no time to check it in depth, the structure looks good but you did not implement `generateFilter`. As far as I remember, it should run, but beware that some functionalities will be broken.

dgerod, 2017-10-20:
Thanks a lot for you answer.

So, I would try to add `generateFilter`? But, what is it? Why it is needed? Could I copy the one from 
`POPFPlanParser`?

A last question, what are Esterel plans? I am just curious :-)

alessio-cpt, 2017-10-20:
I am not in the original developers team, so take the following words with grain of salt ;)

`generateFilters` is used to generate filters every time a new plan is created. You can think of filters as safety sub-routines: if your knowledge base gets updated (e.g. by perception) with some fact which is in contrast with the plan being executed, a notification is sent from the KB to the planning system. This is explained better in their original [paper](https://www.aaai.org/ocs/index.php/ICAPS/ICAPS15/paper/viewFile/10619/10379rl).

If I remember well,  `generateFilters` is called only at the end of the plan parsing, so if you don't call it, you just don't use filters, since no filter are added to the KB. I am not sure this has consequences beyond that, but I do not think so.

If you want to use them, you may try to just copy their function. Make sure you correctly add `filter_objects` in `processPDDLParameters`.

To select a different planner and parser you should refer to the original [wiki page](http://kcl-planning.github.io/ROSPlan/documentation/replace_planner.html). Once everything is set (planner execution command, parser, etc...), I modified this line in `PlanningSystem` constructor from: 

```cpp
PlanningSystem::PlanningSystem(ros::NodeHandle& nh)
		: system_status(READY),
		  plan_parser(new POPFEsterelPlanParser(nh)),
		  plan_server(new actionlib::SimpleActionServer<rosplan_dispatch_msgs::PlanAction>(nh_, "/kcl_rosplan/start_planning", boost::bind(&PlanningSystem::runPlanningServerAction, this, _1), false))
	{
```

To:

```cpp
PlanningSystem::PlanningSystem(ros::NodeHandle& nh)
		: system_status(READY),
		  plan_parser(new YourPlanParser(nh)),
		  plan_server(new actionlib::SimpleActionServer<rosplan_dispatch_msgs::PlanAction>(nh_, "/kcl_rosplan/start_planning", boost::bind(&PlanningSystem::runPlanningServerAction, this, _1), false))
	{
```
You may need some other small tweaks in `PlanningSystem::runPlanner()`.

So no replacing required, add your class and initialize the planning system accordingly. 

About Esterel Plans, you can check this [page](http://kcl-planning.github.io/ROSPlan/documentation/planning_system_plans.html#el) of the documentation . I am not using them right now so I cannot help you much.

Good luck!

m312z, 2017-10-20:
Hello both,

Thanks for your answer Alessio!

The code looks fine for me also.

The `generateFilter` method is for creating a set of objects and propositions which will be monitored by the Knowledge Base. Should one of these change (is added or removed from) the KB, then a notification is sent to the Planning System to trigger a replan. This functionality is not essential.

In fact, anything below the "**// prepare ... for the knowledge filter**" comments can be deleted from the plan parser.

You are right -- to use the new plan parser you must change the source code where the plan parser is created. (line 17 of PlanningSystem.cpp)

The devel branch contains a new version of ROSPlan. Everything is separated into a distinct ROS node. For example, the plan parser subsrices to a "plan<string>" topic, and publishes a "plan<data>" topic. In this situation, everything can be connected up via topics in the launch file. However, there is still a couple of things to disentangle before merging it with master.

Moreover, it seems like a good time to check everything is working on 16.04/Luna.

Michael

alessio-cpt, 2017-10-20:
> The devel branch contains a new version of ROSPlan. Everything is separated into a distinct ROS node. For example, the plan parser subsrices to a "plan" topic, and publishes a "plan" topic. In this situation, everything can be connected up via topics in the launch file. However, there is still a couple of things to disentangle before merging it with master.

This is very good, I'll check it out. Thanks!

m312z, 2017-10-20:
Hello both,

Thanks again for your in-depth reply Alessio! You are correct.

The Esterel plan is the structure used with the EsterelPlanDispatcher. This extends the simple action list to a more complex structure, allowing for branching and looping plans.

Michael

m312z, 2017-10-20:
Hello Alessio,

I just pushed the rest of what I have to the devel branch. You can check the changes using the "test.launch" launch file in the demos package.

You should get a node configuration something like this:
[image]

The problem generation, planner interface, plan parser, are all separate nodes. They were linked together in the launch file, so that triggering the problem generation will start the whole process.

PlanDispatch just needs extracted from PlanningSystem.cpp now.

Michael

dgerod, 2017-10-22:
Hi both,

According to your suggestion:
```
PlanningSystem::PlanningSystem(ros::NodeHandle& nh)
		: system_status(READY),
		  plan_parser(new YourPlanParser(nh)),
		  plan_server(new actionlib::SimpleActionServer<rosplan_dispatch_msgs::PlanAction>(nh_, "/kcl_rosplan/start_planning", boost::bind(&PlanningSystem::runPlanningServerAction, this, _1), false))
	{
```

I have to replace `POPFEsterelPlanParser` by my `FFPlanParser`, but this parser is attached to the `EsterelPlanDispatcher`.
```
plan_dispatcher = new EsterelPlanDispatcher(*dynamic_cast<POPFEsterelPlanParser*>(plan_parser));
```

However, my parser is following original  `POPFPlanParser` instead of  `POPFEsterelPlanParser`. That means, I cannot attach it to the 'EsterelPlanDispatcher'  so I use `SimplePlanDispatcher` instead of 'EsterelPlanDispatcher'.
```
plan_dispatcher = new SimplePlanDispatcher();
```

Is my assumption correct? 

Finally, I have modified the `PlanningSystem` to be able to select different planners when the node starts without recompiling the package. The selection is done using the "planner_command" parameter and I added a new parameter named as "planner_path". It is a temporary solution as in a previous comment is explained that you are working on a solution. 

You could find the modifications in my [develop](https://github.com/dgerod/ROSPlan/commit/9ab4a843cb8f243303866bbd00ea04af5ff32117) of my repository.

Thanks for your support!

alessio-cpt, 2017-10-22:
Yes, your assumption looks correct.

closed 2017-10-28 by dgerod