KCL-Planning/ROSPlan #63: Python Components

Archetype90, 2017-11-13:
Are there any examples of components that are implemented using Python as opposed to C++?

m312z, 2017-11-13:
Hello,

I found one!

It is the interface between PDDL messages:
examine_panel
observe_inspection_point
recalibrate_arm
... etc
On an AUV.

It is not the cleanest code, but should be enough to serve as an example.

https://bitbucket.org/udg_cirs/udg_pandora/src/6b08e2ba9b1f53fdc7d88e8cb627962537f034e6/udg_planning_interface/src/udg_planning_interface.py?at=master&fileviewer=file-view-default

Michael

dgerod, 2017-11-17:
See https://github.com/dgerod/rosplan_pytools, last functionality on in the develop branch. It is based on rosplan_interface repository but with some improvements.

I could provide you an example if you want.

Archetype90, 2017-11-17:
Oh this is fantastic, thank you both so much.

I especially love the interface that you created @dgerod. Something like that is exactly what I was hoping for. If you have any example that you would like to share, I would love to see it. But, the simple documentation that you provided in your ReadMe seems more then sufficient to get me off the ground. 

I will leave this open for now (but please feel free to close) just in case there is anymore feedback.

dgerod, 2017-11-18:
Good, glad that you like :-)

Please, see and example below that I use in my robot to pick an object. In this code I am using a set of skills that my robot is publishing.

```
import rospy
import rosplan_interface.action_interface as planner
import rosplan_interface.kb_interface as kb

import tf
from geometry_msgs.msg import Pose
from rosplan_anchoring.robot.skills import MoveArmSkill, GraspSkill

class PickUpAction(planner.Action):
    name = 'pickup_object'

    def start(self, **kwargs):
   
        container_name = kwargs['c']
        object_name = kwargs['o']

        # 1. Publish feedback to ROSPlan (enabled)
        # 2. Gathering knowledge from KB
        # 3. Interact with robot
        # 4. Update knowledge base
        # 5. Publish feedback to ROSPlan (achieved)
  
        p1 = kb.get_instance(None, container_name, Pose._type)
        target_pose = [p1.pose.position.x, p1.pose.position.y, p1.pose.position.z, 0, 0, 0]

        MoveArmSkill().move_to_pose(target_pose, "/map")
        GraspSkill().grab_object()
        MoveArmSkill().move_to_home()

        self.report_success()
        return True
```
I would like to clarify that I have not create the original source code, I have only updated it to make it more easy to use and added some functionality. It is a fork of` [yochan-lab/rosplan_interface](https://github.com/yochan-lab/rosplan_interface). 

The master branch of [dgerod/rosplan_pytools](https://github.com/dgerod/rosplan_pytools) is linked to the master of _yochan-lab/rosplan_interface_, several of my previous commit are already integrated in _rosplan_interface_ master but no last ones, and I don't know if they will be. Therefore, my advice is that you use the develop branch of _dgerod/rosplan_pytools_. 

If you have any question, let me know in _dgerod/rosplan_pytools_ project.

closed 2018-01-02 by Archetype90