KCL-Planning/ROSPlan #66: Help - dispatch_concurrent, dispatch_on_completion

cheerudeep, 2017-12-04:
I have durative-actions in domain file. Problem is controlling two or more drones to achieve a end goal. After running the planner. Some actions have the dispatch_time I want this actions to be published at the same time, So I can execute them in parallel. 

I tried using the dispatch_concurrent & dispatch_on_completion params to achieve this as mentioned in the Wiki Page f. Planning System. But what ever states (true or false) I set these two params I see no difference. I assume they are not being read by the dispatcher. 

Would be great if someone could help me out. 

Here is by call to the Planning system in Launch file 

```
	<!-- planning system -->
	<node name="rosplan_planning_system" pkg="rosplan_planning_system" type="planner" respawn="false" output="screen">
		<!-- directory for generated files -->
		<param name="data_path" value="$(find rosplan_planning_system)/common/" />
		<param name="problem_path" value="$(find rosplan_planning_system)/common/problem.pddl" />
		<param name="strl_file_path" value="$(find rosplan_planning_system)/common/plan.strl" />

		<!-- to run the planner -->
		<param name="planner_command" value="timeout 10 $(find rosplan_planning_system)/common/bin/popf -n -b DOMAIN PROBLEM" /> 

		<param name="generate_default_problem" value="true" />
		<param name="max_dispatch_attempts" value="10" />
		<param name="dispatch_on_completion" value="false"/>
		<param name="dispatch_concurrent" value="false"/>
		
	</node>
```

m312z, 2017-12-19:
Hello,

The wiki is a bit outdated now -- all the documentation here is more current:
http://kcl-planning.github.io/ROSPlan/documentation/

We have been working on this kind of dispatch just recently. I think that you will be able to find the functionality you are looking for in the development (devel) branch rather than the main branch. In particular the PDDLEsterelPlanParser and the EsterelPlanDispatcher nodes.

You could try switching to the development branch, which is less stable, but has far better features, or if you prefer, copy the code that you require and recompile.

We are aiming to merge everything into master ASAP.

Michael

Archetype90, 2018-01-31:
Hi @m312z 

Do you have a time frame for merging the develop branch into master? Thanks!

m312z, 2018-02-02:
Hello,

The Turtlebot demo from the readme now works in the devel branch using the same launch/run commands as the master branch. Hopefully this means that disruption will be minimised.

There are a couple of other changes and fixes to merge and then test. Then I'm hoping to merge next week (on the 8/9th Feb).

Michael