KCL-Planning/ROSPlan #67: RPSimpleMapServer

morris0427, 2017-12-11:
Hello,

I'm trying to play with the turtlebot code using waypoints I manually specify and the SimpleMapServer rather than the PRM server. I want to visualize the graph in rviz.  I generate a PDDL planning problem, which gets parsed correctly. But after the action is enabled by the planning system I get the following error:

[ERROR] ... The goal pose passed to this planner must be in the map frame. It is instead in the world frame. 

I've been studying the RPSimpleMapServer and Visualization codes and notice that in the code that publishes the waypoint markers, in the simple map server the frame_id is a fixed world frame, whereas in the PRM visualization code the frame id is set to 'map'. I expect that's partly why the PRM visualization works and the simple map visualizer does not, but I wonder if you have any thoughts.

Thanks,
Robert Morris

m312z, 2017-12-19:
I've just looked over the code:
- In the roadmap (PRM) server, the frame is taken from the occupancy map.
- In the PRM visualisation the frame is set to "map"
- in the simple map server the frame is set by a launch parameter (fixed_frame).
- in the simple map visualisation the frame is the same variable.

If you are using simple map server and have not done so already, you might want to add the parameter to the launch file, such as:
<node ... type="rpsimplemapServer" ...>
    ...
    <param name="fixed_frame" value="map" />
</node>

Then in rviz when visualising the waypoints, make sure the fixed_frame is set to the same value ("map" in the above code, but whatever is the fixed frame of your system.)

Hope this helps,
Michael

morris0427, 2018-01-08:
Michael,

Thanks for your note.

In rosplan_roadmap_server.launch the type of the roadmap server is set to rpsimplemapServer as you suggested below. I defined a simple problem where kenny is at wp0 and needs to go to wp2 through wp1. All of the waypoints are defined in a file. The file is read properly as far as I can see. I have a problem file and simple bash script file that defines the waypoints and goals of the problem. When ROSPlan is run the problem file that is generated is the one that I want, and the planner comes up with the right plan in the plan.pddl file. So nothing seems to be wrong on the planning side.

So the problem apparently is in the visualization. When I run the bash script it is read without error but in the terminal in which I run the turtlebot launch file I get the error “The goal pose passed to the planner must be in the map frame. It is instead in the world frame”. In rviz I set the fixed_frame to “map” and get the same problem.

If instead of the simple map Server I set the roadmap server to  rproadmapServer in the roadmap server launch file the demo runs correctly.

Robert

morris0427, 2018-01-08:
Michael,

In RPSimpleMapServer.cpp I changed the parameter of fixed_frame from ‘world’ to ‘map’ (in main) and that seems to fix my problem.

Thanks,
Robert

closed 2018-03-16 by m312z