KCL-Planning/ROSPlan #69: Wiki Removed?

Archetype90, 2018-01-03:
As can be seen from the Wiki History:

https://github.com/KCL-Planning/ROSPlan/wiki/_history

Sections A. through F. were removed 15 days ago. Would it be possible to restore these? Thanks!

YanielCarreno, 2018-01-06:
Yes!! I noticed the same. I was wondering if it is possible to have this content online again. Thanks!!

moriarty, 2018-01-06:
The GitHub wikis are in git repos themselves, so the changes can be reverted.
```git clone https://github.com/KCL-Planning/ROSPlan.wiki.git```

@m312z was there a reason they were destroyed, or just a mistake?

[image]

m312z, 2018-01-10:
Hello both,

The wiki was getting disastrously out of date. All of the information that was removed from there, plus more is available on the ROSPlan website here:

http://kcl-planning.github.io/ROSPlan/documentation

Michael

Archetype90, 2018-01-12:
@m312z There are some details missing from the ROSPlan web page. For example, the old Getting Started page from the Wiki described the exact steps to get the TurtleBot plan running. This was phenomenally helpful for me getting started and I am sure other new users would love that guide as well.

Archetype90, 2018-01-16:
Just to follow up on this with an example;

http://kcl-planning.github.io/ROSPlan/documentation/add_action.html

The add an action documentation from the official site actually references the Wiki for additional information for creating an action.

m312z, 2018-01-26:
Hello again,

I've just updated the "getting started" steps to match the updated code. Right now it is in the README on the devel branch, which will soon be merged into master. Before that happens there is some more documentation to update. (This includes adding actions).

Michael

Archetype90, 2018-01-31:
Thanks very much for the update @m312z . Closing the issue.

closed 2018-01-31 by Archetype90