KCL-Planning/ROSPlan #70: Domain File Does Not Exist

Archetype90, 2018-01-12:
There seems to be an issue with the Planning System outputting log messages that the Domain file does not exist. Here is my situation..

I have a PDDL domain, which I found in the following repo:

https://bitbucket.org/planning-researchers/classical-domains/src/208a850d2ff2a27068329ad578ad99af9ec7e5c5/classical/tsp/domain.pddl?at=master&fileviewer=file-view-default

My goal was to find as simple of a PDDL example as possible and build a custom component around that PDDL. 

The PDDL domain is as follows: 

```
(define (domain tsp)

(:requirements :strips :typing :fluents :disjunctive-preconditions :durative-actions)

  (:types 
  	location)

  (:predicates
     (at ?x - location)
     (visited ?y - location))

  (:action move
	:parameters (?x ?y)
	:precondition (at ?x)
	:effect (and (at ?y) (visited ?y) (not (at ?x))))
)
```

After launching the Planning System and all other components (using a modified interfaced_planning_system.launch that replaces the domain and component) everything seems to be fine. I then rush a bash script to populate my knowledge base and start the planning system. 

After running the bash script, I see the following log messages: 

```
[ INFO] [1515784983.314983688]: KCL: (KB) Adding instance (location, 1)
[ INFO] [1515784983.876731650]: KCL: (KB) Adding instance (location, 2)
[ INFO] [1515784985.185522507]: KCL: (KB) Adding domain attribute (at location=1)
[ INFO] [1515784985.883142831]: KCL: (KB) Adding mission goal ( visited=1)
[ INFO] [1515784986.524591316]: KCL: (PS) Parsing domain: /home/ros/ros_plan/src/rosplan/rosplan_demos/common/domain_tdl.pddl.
[ INFO] [1515784986.524948030]: KCL: (PS) File does exist: /home/ros/ros_plan/src/rosplan/rosplan_demos/common/domain_tdl.pddl.
[rosplan_planning_system-4] process has died [pid 65091, exit code -11, cmd /home/ros/ros_plan/devel/lib/rosplan_planning_system/planner __name:=rosplan_planning_system __log:=/home/ros/.ros/log/7d2364d4-f6ff-11e7-aaf0-000c2948cd07/rosplan_planning_system-4.log].
log file: /home/ros/.ros/log/7d2364d4-f6ff-11e7-aaf0-000c2948cd07/rosplan_planning_system-4*.log
```

Now, this is not true. First, if I copy/paste the "missing" file location, it opens with no problems. And the log file that it specifies (rosplan_planning_system-4*.log) does not exist at all. In fact, that folder does not contain any rosplan_planning_system logs at all. 

Now, I would find the errors above odd, but this is where things get really weird. If I then replace the PDDL in the domain file with the following:

```
(define (domain message_demo)

(:requirements :strips :typing :fluents :disjunctive-preconditions :durative-actions)

(:types
	message
	robot
)

(:predicates
	(said ?s - message)
	(robot_at ?v - robot ?m - message)
	(connected ?curr ?next - message)
)

;; Move between any two waypoints, avoiding terrain
(:durative-action say_message
	:parameters (?v - robot ?curr ?next - message)
	:duration ( = ?duration 10)
	:condition (and
		(at start (robot_at ?v ?curr))
	:effect (and
		(at end (said ?next))
		(at start (not (robot_at ?v ?curr)))
		(at end (robot_at ?v ?next))
)
)
```

I then start the same launch file and bash script. After running the bash script I get the following output:

```
[ INFO] [1515785193.941940592]: KCL: (KB) Adding instance (location, 1)
[ INFO] [1515785195.168009198]: KCL: (KB) Adding instance (location, 2)
[ INFO] [1515785196.239814408]: KCL: (KB) Adding domain attribute (at location=1)
[ INFO] [1515785197.345461896]: KCL: (KB) Adding mission goal ( visited=1)
[ INFO] [1515785197.844691363]: KCL: (PS) Parsing domain: /home/ros/ros_plan/src/rosplan/rosplan_demos/common/domain_tdl.pddl.
[ INFO] [1515785197.845050084]: KCL: (PS) File does exist: /home/ros/ros_plan/src/rosplan/rosplan_demos/common/domain_tdl.pddl.
[ INFO] [1515785197.845683922]: KCL: (PS) Finished parsing domain: /home/ros/ros_plan/src/rosplan/rosplan_demos/common/domain_tdl.pddl.
[ INFO] [1515785197.845969577]: KCL: (PS) Fetching objects
```

As you can see, it shows the same "file does not exist" log message, but manages to finish parsing the domain file anyway. The system then goes on to generate a problem file and attempt to plan. 

**In both situations the "file does not exist" error is not accurate. The file does exist.** This is the issue that I am submitting.

What I would like to understand, though, is why the system in some cases says the file does not exist, but parses it anyway, and in other it says the file does not exist, crashes, and does not produce a log message?

m312z, 2018-01-12:
Hello there,

It looks like the output you show here says: "File does exist". This would mean that in both cases the file is found.

I think that the crash is due to missing types in the action parameters:
> :parameters (?x ?y)

Try:
> :parameters (?x ?y - location)

If that works, then it looks like there is a bug in the domain parser -- not checking for the existence of types before trying to read them. Hopefully you are able to use typed parameters and avoid this issue for now.

Michael

m312z, 2018-01-13:
@b-dorian

Archetype90, 2018-01-15:
Thanks a lot for the reply @m312z. What a silly mistake on my behalf. Your solution did work and removed the error. I have updated the question to more accurately reflect the useful information that you have provided, and not my mistake.

Thank you again for your reply!

closed 2018-07-31 by Archetype90