KCL-Planning/ROSPlan #74: In case of ROSPlan I want to clarify my doubt regarding knowledge base and mongodb.

yashu88, 2018-02-14:
In case of ROSPlan I want to clarify my doubt regarding knowledge base and mongodb.

1. Information such as "instances","facts" are stored in Standard Template Library and obtained from STL? 
or 
2. Knowledge management service with the help of .srv and .msg file just import and export information between nodes? 

Then what is the use of MongoDB here? Is problem generation or plan dispatch file uses message_store.0 file contents in decision making?

As per my understanding KB with help of .msg and .srv access domain.pddl information and passes to required nodes & src files. problem.pddl files generation is not fully automatic but depends on code mentioned in rosplaning_system/src directory source code. RPSimpleMapServer.cpp files contains source code related to mongodb message_store file. Here mongodb used to stored geometric information related to waypoints. Every time KB is populated with the help of shell script. It's my assumption about this source code. Kindly clear my doubts. 

Kindly help me out on this.

dgerod, 2018-02-15:
MongoDB is the scene db while data stored in the node is the knowledge base.

In the scene db you are storing isubsymbolic information as a pose, a laser scan, etc and in KB you are storing symbols. Information in the scene db is linked with a instance of the KB by name. However, this management must be done by you as the KB node does not do that.

For example, if in an action you add an object to the KB, you should add a pose to the scene db with the same name. Next time you need the pose of this object you request to the scene db.

yashu88, 2018-02-19:
Yes as you mentioned in your answer MongoDB is the scene db while data stored in the node is the knowledge base. My assumption was the same because during my source code study in launch file I found out there are two nodes one is rosplan_knowledge_base (knowledge base) and other is rosplan_scene_database ((MongoDB)

	****

	<node name="rosplan_knowledge_base" pkg="rosplan_knowledge_base" type="knowledgeBase" respawn="false" output="screen">

	****

	<node name="rosplan_scene_database" pkg="mongodb_store" type="mongodb_server.py" respawn="false" output="log">
	<node name="rosplan_scene_message_store" pkg="mongodb_store" type="message_store_node.py" respawn="false" output="log" />

But my main question in above post is I couldn't find out details regarding
Information in the message_store.0 (MongoDB) is linked with a instance of the KB by name. In short I want to find out using problem.pddl and domain.pddl ROSPlan planner generate plan.pddl whether planner also import information from message_store.0 for generating plan. 

I got confused bcoz I found out script file mentioned in rosplan_demos every time populate knowedge base with random waypoint generation. I am sorry for taking your time.

fmrico, 2018-02-19:
I am using ROSPlan onboard of the Pepper Robot. I have a version without MongoDB because is hard to cross compile, and it is not justified its practical use. My node in charge of symbolic predicates related to waypoints also have a service that translates symbolic locations to geometry_msgs/Pose.

I prefer to remove MongoDB from the core of ROSPlan.

yashu88, 2018-02-19:
Sir did you remove MongodB by modifying source code or did you use some other version of ROSPlan (without MongodB)?

If so kindly let me know the version (without DB) you are using. Thanks.

fmrico, 2018-02-19:
Only deactivating MongoDB in launchers. This does not affect affect to the core functionality of ROSPlan, as @dgerod said. Of course, you have to do several changes in source code of the nodes involved in demos, or develop your own code for actions and predicate generators without MongoDB.

Do a `rgrep` looking for message_store ;)

oscar-lima, 2019-05-03:
rosplan_demos are now moved to: https://github.com/KCL-Planning/rosplan_demos
and there are plans to remove mongoDB from the rosplan_demos in favor of storing the waypoints in the ROS parameter server. By moving the demos, the core implementation of ROSPlan is now independent from mongoDB.

closed 2019-05-03 by oscar-lima