KCL-Planning/ROSPlan #80: Use ontology to generate more accurate problem instance

yashu88, 2018-02-26:
Is it possible to combined domain model data along with sensor and geometrical data in form of ontology to generate more accurate problem instance?

dgerod, 2018-03-02:
It seems that some people did as a side version of ROSPlan: https://github.com/EmaroLab/OWL-ROSPlan

yashu88, 2018-03-02:
Thanks I will check fork version of  KCL-Planning ROSPlan to learn about How to utilize ontology generation. Actually in KCL-Planning ROSPlan (source code) also I have seen some lines regarding ontology and also I found in respective research article that ontology concept have been used by researcher during case study. So I was curious. Anyways Thank you.

oscar-lima, 2019-05-03:
ROSPlan currently lacks support for ontologies, you might want to check http://www.knowrob.org/ for that. Some owl to pddl parsers were developed in the research community (not sure if they are open source), it would be nice to explore this options as research directions.