KCL-Planning/ROSPlan #82: [Q&A] SimplePlanDispatcher

dgerod, 2018-03-04:
I am using the FF planner with some PDDL files and it solves the plan, see below the output:
```
ff: found legal plan as follows

step    0: MOVE_TO_PLACE B1 C1
        1: FIND_OBJECT C1 V1
        2: FIND_OBJECT C1 V2
        3: FIND_OBJECT C1 V3
        4: FIND_OBJECT C1 V4
        5: PICKUP_OBJECT C1 V1
        6: MOVE_TO_PLACE C1 B1
        7: DROP_OBJECT B1 V1
        8: MOVE_TO_PLACE B1 C1
        9: PICKUP_OBJECT C1 V2
       10: MOVE_TO_PLACE C1 B1
       11: DROP_OBJECT B1 V2
       12: MOVE_TO_PLACE B1 C1
       13: PICKUP_OBJECT C1 V3
       14: MOVE_TO_PLACE C1 B1
       15: DROP_OBJECT B1 V3
       16: MOVE_TO_PLACE B1 C1
       17: PICKUP_OBJECT C1 V4
       18: MOVE_TO_PLACE C1 B1
       19: DROP_OBJECT B1 V4
```
However, when the `SimpleDispatchPlanner` executed the plan after being correctly parsed, it says that preconditions are not meet for 1st FIND_OBJECT action, see below the messages provided by `rosplan_planning_system`. I have removed some messages provided by `FFPlanParser` after second FIND_OBJECT action because they are not interesting in current situation.
```
[ INFO] [1520146202.663431091]: KCL: (PS)(FFPlanParser) Generate action - ID: 1, Name: move_to_place
[ INFO] [1520146202.663469716]: KCL: (PS)(FFPlanParser) Generate action - ID: 2, Name: find_object
[ INFO] [1520146202.663512507]: KCL: (PS)(FFPlanParser) Generate action - ID: 3, Name: find_object
...

[ INFO] [1520146202.664302759]: KCL: (PS) (problem.pddl) Clean and update knowledge filter
[ INFO] [1520146202.664401452]: KCL: (PS)(SPD) Dispatching plan
[ERROR] [1520146202.664437240]: KCL: (PS)(SPD) Message action_id [2] does not meet expected [1]
[ INFO] [1520146202.665697843]: KCL: (PS)(SPD)        [robot-at]
[ INFO] [1520146202.665720823]: KCL: (PS)(SPD)        [localised]
[ INFO] [1520146202.665740002]: KCL: (PS)(SPD) Preconditions not achieved [2, find_object]
[ INFO] [1520146202.665776138]: KCL: (PS) (problem.pddl) The plan failed!
```
The point is that I have some a negative precondition in FIND_OBJECT actions, see below. 
```
(:action find_object
	:parameters (?c - container ?o - object)
	:precondition
	    (and (robot-at ?c)
	         (not (localised ?o)))
	:effect
	    (and (robot-at ?c)
	         (localised ?o)
		 (object-at ?o ?c))
)
```
I think that `SimpleDispatchPlanner` wants that I set explicitly these one negated when the FF planner does not need them. The problem file that I am passing to the FF planner is: 
```
(define (problem organize_boxes_task)
(:domain organize_boxes)
(:objects
    b1 - box
    c1 - container
    v1 v2 v3 v4 - videotape
)
(:init
    (free-tool)
    (object-at v1 c1)
    (object-at v2 c1)
    (object-at v3 c1)
    (object-at v4 c1)
    (robot-at b1)
)
(:goal (and
    (object-at v1 b1)
    (object-at v2 b1)
    (object-at v3 b1)
    (object-at v4 b1)
)))
```
Why `SimplePlanDispatcher` is doing this? Is this behavior correct? In my opinion it should not be as the FF planner is able to solve this situation... or the FF planner behavior is not.  And How I can fix this situation? 

Thanks.

dgerod, 2018-03-04:
It seems that the system does not work when a PDDL domain file is using a negate precondition, I have found the following comment in `parsePrecondition()` of `PlanningEnvironment`: 
>  TODO: test negative precondition support.

Am I right? How to solve it?

m312z, 2018-03-06:
(1) Looking through the plan dispatcher code, negative preconditions don't appear to be supported at all. The code you mention does not look at the "neg_goal" VAL type, only "simple_goal".

(2) In the Knowledge Base, a negative precondition will not be detected if the positive fact is not in the KB. However, adding the negative precondition with "is_negative=true" in the KnowlegeItem should work.

To solve the problem in the short term, some code must be added to PlanningEnvironment to iterate through the neg_goal list, and then explicitly adding the negative facts to the KB should work.

I'll leave this issue open and post an update when there is a complete fix.

dgerod, 2018-03-06:
Ok, thanks. I will check during the weekend.