KCL-Planning/ROSPlan #85: Problem while using rosplan_interface_COLA2 interface

yashu88, 2018-03-09:
I have included auv_msgs, cola2_core and cola2_s2 packages in my catkin/ src folder along with rosplan_interface_COLA2. But I got error "there is no such file "WorldWaypointReqAction.h" in cola2_control". I check and found out this file belongs to cola2_msgs package. so I just replace cola2_control with cola2_msgs. 

#include "cola2_control/WorldWaypointReqAction.h"  change to                                                            #include "cola2_msgs/WorldWaypointReqAction.h"

After this I have successfully compiled. 
Actually I want to try something like this
https://www.youtube.com/watch?v=qD2fHkX-XPU (ROSPlan SPARUS II Simulation) for this I have modify sparus.launch file by adding 

	
	<include file="$(find cola2_s2)/launch/sim_s2_start.launch" >
         </include>
 
Also I have created domain.pddl file similar to domain_turtlebot_demo.pddl. It begins the simulation, rviz visualisation, and ROSPlan nodes 

Then run sparus_explore.bash script. 
But still I didn't get expected result. I might be wrong some where. Kindly help me out. Thank you.
[image]

m312z, 2018-03-09:
What is the result that is expected, and what is the actual result?

The output in the screenshot looks correct, with the exception of the warning which I have not seen before. After the goto_waypoint is dispatched, the next line I would expect is from the rosplan cola2 interface:
  ROS_INFO("KCL: (Cola2) action recieved"); // (From line 29 of RPCola2.cpp)
This node is subscribed to the action dispatch, and the first thing it does is print confirmation that it has received an action.

If you do not see this line, then make sure that the cola2 interface subscription and the action dispatch are on the same topic.

yashu88, 2018-03-09:
Exactly I didn't receive "action received",  Instead I have received warning " /computer_logger: unable to get data: <type 'exceptions.OSError'> . I didn't see way-point generation on rviz visualisation. 

I will check  whether cola2 interface subscription and action dispatch are on the same topic or not.  I also need to look into sparus.launch file. 

Thank you for help.

yashu88, 2018-03-15:
Hello,

I have fix the problem regarding "subscription and the action dispatch are on the same topic" as per your suggestion. 

Now after "goto_way-point dispatch" next thing it print is "action received".

But I am still receiving those warnings

[WARN] [1521100524.007742]: /computer_logger: unable to get data: <type 'exceptions.OSError'>

I think they belongs to cola2_control package. Because even though rosplan shows action enable in rqt interface but there is no action perform on rov (no surge, sway movement) .

Also unlike turlebot here I can not see way-point marking on Rviz.

As per my understanding cola2 interface unable to establish link between rosplan and cola2_control package that might be the reason behind those warnings.

Sorry for taking your time. Hope you can guide me in right direction.

Thank you.

[image]