KCL-Planning/ROSPlan #94: [Q&A] Could 'forall' be used in goals?

dgerod, 2018-04-22:
I would like to define a goal using 'forall', see below. I have created domain and problem PDDL files to be used with Contingent-FF planner, and it solves the problem correctly when I call it from the terminal. 
```
(:goal (forall (?o - object)
            (and (object-at ?o box1))
))
```
So, my question is how I have to pass this type of goal to the knowledge base? 

Thanks in advance.

m312z, 2018-07-10:
Currently the Knowledge Base interface does not support forall goals. It is possible to send this goal as a conjunction; e.g.
```
(:goal (and
  (obbject-at obj1 box1)
  (obbject-at obj2 box2)
```
By first fetching the object instances and then using a single KnowledgeUpdateServiceArray.

A more elegant solution goes onto the development list...

dgerod, 2018-07-11:
OK, thanks for your answers.

dr-steveb, 2019-03-12:
Any future plans to implement forall or when?