KCL-Planning/ROSPlan #100: Turtlebot not moving with waypoints

sharon50270, 2018-05-31:
Hi,
I followed the .readme try to have the turtlebot do waypoints. After running turtlebot.launch I rosran turtlebot_explore.bash, the output on the first terminal are all true and waypoints were shown in rviz but the Turtlebot was not moving at all. Is there anything else I should do? Thank you!

yashu88, 2018-06-29:
Source your rosplan under catkin workplace using source devel/setup.sh. It might solve your problem.

m312z, 2018-07-10:
In the script, there are the following lines:
```
echo "Calling problem generator.";
rosservice call /rosplan_problem_interface/problem_generation_server;

echo "Calling planner interface.";
rosservice call /rosplan_planner_interface/planning_server;

echo "Calling plan parser.";
rosservice call /rosplan_parsing_interface/parse_plan;

echo "Calling plan dispatcher.";
rosservice call /rosplan_plan_dispatcher/dispatch_plan;
```
What output do you see in the script window?

You might be using a different/older version of that script which does not automatically start the dispatch.

sharon50270, 2018-08-16:
It's been solved by sourcing this workspace exclusively. I sourced after installation with other workspace, which led to this issue. Thank you guys!

closed 2018-08-16 by sharon50270