KCL-Planning/ROSPlan #107: Regarding persistent knowledge base!

yashu88, 2018-06-29:
I want to join MongoDB to ontology for giving global memory effect. 

Before that I tried to insert PDDL model in to MongoDB along with geometrical information of waypoint. Then I queried my MongoDB and generate my own new domain.pddl file.

 In current scenario this new domain.pddl file is generated using parsed data from original file. In future I want to replace that parsed data with data from ontologies. 

But I am facing some issue while generating domain model. Due to usage of srv/msg while inserting data into MongoDB while querying it access all information from start. It didn't access particular information related to that particular topic. Also I want to know like in interface_mapping file you have used delete query to clear waypoints from MongoDB , If similar method I have used It didn't work and give some kind of warnings.
[image]

Kindly help me out. Thanks. 

[image]

m312z, 2018-07-10:
Hello,

I'm not exactly sure I understand what you are describing. From the error messages in the second screenshot, it appears that there is no supplied objectID.

In general, you can find information on how to use the ROS mongodb_store here:
http://wiki.ros.org/mongodb_store

Michael

yashu88, 2018-07-11:
Hello,

Thank you for kind reply.

Actually I have modified source code and generate domain file. Now it's almost similar to original one. Still some changes has to be done.

Now I am working on delete query.

[image]