KCL-Planning/ROSPlan #110: undefined reference to mongo::BSONObj::jsonString

ysl208, 2018-07-13:
Hi,

When i run catkin build I get the following errors for both rosplan_interface_movebase and rosplan_interface_mapping 

> undefined reference to `mongo::BSONObj::jsonString(mongo::JsonStringFormat, int, bool) const'
> collect2: error: ld returned 1 exit status

I have mongodb_store installed. 

> ros-indigo-mongodb-store/trusty,now 0.4.2-0trusty-20180603-233334-0800 amd64 [installed]

So not sure what the cause is.
I appreciate your help.

m312z, 2018-07-17:
Is there any more information in the error message? (e.g. which line is causing the error)

Michael

ysl208, 2018-07-17:
Hi Michael, thanks for getting back at me. Added the entire error message. in the original post