KCL-Planning/ROSPlan #115: concurrent plan execution for multiple robots are not working in ROSplan

Etns, 2018-07-24:
Hi, 

I've encountered a problem while executing parallel actions.
As shown below, concurrently dispatched actions are executed sequentially.

Am I missing some points? 

Any comments would be appreciated very much.
 
[image]

m312z, 2018-07-24:
Hello there,

From your diagram I am assuming that the three forklift actions are dispatched at 0.000, but then they are executed in sequence?

Take a look at the code that implements the execution of the forklift action (that is subscribed to the PDDL action message) perhaps this code is resolving the actions in sequence, despite there being multiple forklifts?

Best regards,
Michael

Etns, 2018-07-25:
While using Action Interface based on RPSimulatedActionInterface Class,
ros::Rate.sleep() is used to set the duration of action.
Contrary to my expectation, it ends up with not receiving other topics or services,
and am working on to fix this problem.

Am I going to right direction?

Appreciate your help.

m312z, 2018-07-26:
Yes, that's the right direction. The simulated action interface can only serve one action at a time (this is not completely unreasonable, as it resembles real actions such as move_base goals).

However, for the same action on different robots, it makes sense that they run in parallel. Please let me know if you are able to update the action interface to cope with this -- it would be useful to merge in!

Michael

Etns, 2018-07-27:
Sure, willing to share the codes.

YanielCarreno, 2019-05-20:
Hi there,
I found a way to solve this problem. I made some variations in the ActionInterface and it works well for multiple robots implementing the same action concurrently.

oscar-lima, 2019-05-20:
Thanks! Can you please open a pull request and refer this issue from it? We can then provide feedback

SarveshS325, 2019-07-03:
I've run into the same issue. Correct me if I am wrong, concurrent execution of same action by multiple robots needs to be handled by an Action Interface extending RPActionInterface.cpp. My current understanding is that to do this the concreteCallback method of the action interface needs to be run on separate threads for each robot that is supposed to perform the action. Any help regarding this would be much appreciated

YanielCarreno, 2019-07-03:
Hi there, sorry for the delay in my answer, I will open a pull request today with my solution to this problem.
I modified the Action Interface to solve the problem.

m312z, 2019-07-03:
Hello Sarvesh,

The solution will depend a bit upon how the multiple robots are set up (e.g. in separate namespaces, or on different physical machines, etc.).

It can be straightforward:

- Each robot should be running its own action interface. For example, this means for the action "do_action" you will run the action interface on each robot.
- As a parameter of the "do_action" interface, you can set the robot name, or take it from the namespace.
- In the concrete_dispatch method of the "do_action" interface, you can check the robot name matches the action parameter before executing the action.

To run an interface for each robot, make sure the node names do not clash, either:
- give each interface its own node name: e.g. "do_action_interface_robot1".
- put them in the namespaces of each robot, e.g. <group "robot1"> ... </group>

If you need an action that runs on one machine, and in the planning model can be run against itself in parallel an unlimited number of times, then you might need to look into threading. However, if the number of actions is limited (e.g. one for each simulated robot) or the actions are on separate machines (e.g. separate robots/namespaces) then this should suffice.

Michael

SarveshS325, 2019-07-07:
Hi Michael,

Thanks for your reply.

My setup in its current state has a total of 13 actions and 4 robots which are set up in distinct namespaces (This is to be run on one machine since I'm trying to simulate in gazebo). All of the 4 robots may be tasked with  performing any of the said actions at the same time. 
Just to confirm that I understand what you are saying, 

1. Run the action interfaces for the 13 actions under the namespaces of each robot

SarveshS325, 2019-07-07:
Hi Michael,
 Thanks for the reply.

My setup in its current state has 13 actions and 4 robots which are in separate namespaces. Each of the said actions may be dispatched to the robots concurrently.

What I got from comment is that I should run the interfaces for the 13 actions under the namespaces for each robot i.e a total of 52 nodes all of which are subscribed to the dispatch topic.

If I compare the action parameter and robot name within the concreteCallback method then the callback for the appropriate robot will run. Here is where I have a doubt. Doing this will also call concreteCallbacks for the interfaces with the same action name in other namespaces as well (though nothing will happen) and should return a boolean. This will be passed on the feedback topic and will create problems.
A solution could be to compare the action parameter and robot name before calling concreteCallback i.e in the dispatchCallback method in RPActionInterface.

Do I have this right?
Thanks
Sarvesh