KCL-Planning/ROSPlan #118: Time for building Esterel Plan Graph

Samcheok, 2018-07-26:
Hello,

On the model of relatively small toy domain,  
it is observed that compared to the planning time, it takes quite some time to build Esterel plan graph.
For instance, a couple of minutes for planning and an hour to build Esterel graph.

Wondering if there is any other way to turn around this?

Any comments would be appreciated.

Kim Kyeong Seok.

m312z, 2018-07-26:
That is unusual. The process of building the Esterel messages should be very fast, as it only visits each action in the plan once to produce a message, and within that to check for ordering constraints.

If it is taking longer than planning then something different is happening. Can you send a copy of your domain and problem file so that we can find out what?

Michael

Samcheok, 2018-07-27:
Attached are the PDDL files and the plan.

It took 2~3 seconds for planning using POPF; 
building plan graph continues more than an hour. 

Thank you so much for your help!

Kim Kyeong Seok

Samcheok, 2018-07-27:
[pddl.zip](https://github.com/KCL-Planning/ROSPlan/files/2235315/pddl.zip)
Attached are the PDDL files and the plan.

It took 2~3 seconds for planning using POPF; 
building plan graph continues more than an hour. 

Thank you so much for your help!

jqs7328, 2018-08-01:
Hi, 

Strugging with this is problem, we have tested  
our code (posted on the previous reply) with lpg-td and popf:

	               Plan length	Total time	Time for plan gen	   Time for plan graph gen
Lpg-td-1.0	80 actions	103 sec	        7 sec	                           3 min
Popf	                80 actions        207 sec	        4 sec	                           > 5 hr

Additionally, "Mapanalyzer" domain from 2014 IPC  is tested with lpg-td.

                             Plan length	       Total time	        Time for plan gen   Time for plan graph gen
pfile 3-4-2-0-1       23 actions           383 sec              < 1 sec                  < 1 sec             
pfile 15-15-3-1-1  193 actions       3898 sec            91 sec                      20 min

Noticed that plan structure/quality and plan length make a difference on making plan graph.

Any comment would be appreciated very mcuh.

-- jashin