KCL-Planning/ROSPlan #119: RQT GUI gets stuck if I open any plugin.

Rafaelkas, 2018-07-30:
Hi,
I work on ros kinetic, and downloaded the appropriate version of ROSPlan. I face difficulties in using qrt plugins. I have python 2.7 installed.
What should I do?

Error:
RosPluginProvider.load(rosplan_rqt/ROSPlanEsterelPlanViewer) exception raised in __builtin__.__import__(rosplan_rqt.esterel_plan_viewer, [ROSPlanEsterelPlanViewer]):
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 80, in load
    module = __builtin__.__import__(attributes['module_name'], fromlist=[attributes['class_from_class_type']], level=0)
  File "/home/labkinetic/catkin_ws/src/ROSPlan/src/ROSPlan/rosplan_rqt/src/rosplan_rqt/esterel_plan_viewer.py", line 3, in <module>
    from .ROSPlanEsterelPlanViewer import EsterelPlanViewerWidget
  File "/home/labkinetic/catkin_ws/src/ROSPlan/src/ROSPlan/rosplan_rqt/src/rosplan_rqt/ROSPlanEsterelPlanViewer.py", line 22, in <module>
    from python_qt_binding.QtWebKit import QGraphicsWebView
ImportError: No module named QtWebKit

PluginManager._load_plugin() could not load plugin "rosplan_rqt/ROSPlanEsterelPlanViewer":
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 99, in load
    self._load()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 54, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 86, in load
    raise e
ImportError: No module named QtWebKit

m312z, 2018-07-31:
Hello,

I'm neither a python or qt expert. From searching "ImportError: No module named QtWebKit" it looks like there are many problems with that module in qt4 and python2, and also many different solutions. This thread has a few people work through diagnosis which might help you:
https://ubuntuforums.org/showthread.php?t=2253348

Michael

mattGT87, 2018-08-13:
I am using the same environment and was experiencing this same error. I'm also not a python/qt expert but I was able to get the plugin to stop immediately crashing by doing the following in ROSPlanEsterelPlanViewer.py. It seems that some of the qt class names have changed or been rearranged in recent releases, which I think was the root cause of this particular crash. This probably only works for 16.04/kinetic

**Replace (lines 21-28):**
from python_qt_binding.QtSvg import QGraphicsSvgItem
from python_qt_binding.QtWebKit import QGraphicsWebView

if QT_BINDING_VERSION.startswith('4'):
    from python_qt_binding.QtGui import  QWidget, QGraphicsScene, QGraphicsItem, QGraphicsEllipseItem
    from python_qt_binding.QtSvg import  QGraphicsSvgItem, QSvgRenderer
else:
    from python_qt_binding.QtWidgets import QWidget

**With:**
from python_qt_binding.QtWebKitWidgets import QGraphicsWebView
from python_qt_binding.QtWidgets import QWidget, QGraphicsScene
from python_qt_binding.QtSvg import  QGraphicsSvgItem, QSvgRenderer

oscar-lima, 2018-11-19:
I believe this was solved by #112 and #149 so it can be closed as well