KCL-Planning/ROSPlan #130: tutorial_04 - getting stuck

brean, 2018-09-28:
Hi,

I am currently going through the [tutorial](http://kcl-planning.github.io/ROSPlan/documentation/) and I noticed that [tutorial 4](http://kcl-planning.github.io/ROSPlan//tutorials/tutorial_04) does not work as described.
When I run the last command `rosservice call /rosplan_plan_dispatcher/dispatch_plan` it does not only show one action but a lot more
```
[ INFO] [1538127173.175289285]: KCL: (/rosplan_plan_dispatcher) Dispatching plan.
[ INFO] [1538127173.177307618]: KCL: (/rosplan_plan_dispatcher) Dispatching action [0, undock, 8.131856, 10.000000]
[ INFO] [1538127173.178392294]: KCL: (undock) action received
[ INFO] [1538127173.185504191]: KCL: (/rosplan_knowledge_base) Removing domain attribute (docked)
[ INFO] [1538127173.186548846]: KCL: (undock) Action completing with probability 1.000000 and duration 0.000000
[ INFO] [1538127173.187399696]: KCL: (/rosplan_knowledge_base) Adding fact (undocked kenny)
[ INFO] [1538127173.275491759]: KCL: (/rosplan_plan_dispatcher) Feedback received [0, action enabled]
[ INFO] [1538127173.275532283]: KCL: (/rosplan_plan_dispatcher) Feedback received [0, action achieved]
[ INFO] [1538127173.375761433]: KCL: (/rosplan_plan_dispatcher) 0: action undock completed
[ INFO] [1538127173.377914729]: KCL: (/rosplan_plan_dispatcher) Dispatching action [1, localise, 18.132857, 60.000000]
[ INFO] [1538127173.380276865]: KCL: (localise) action received
[ INFO] [1538127173.380391176]: KCL: (localise) Action completing with probability 1.000000 and duration 0.000000
[ INFO] [1538127173.390964001]: KCL: (/rosplan_knowledge_base) Adding fact (localised kenny)
[ INFO] [1538127173.476487526]: KCL: (/rosplan_plan_dispatcher) Feedback received [1, action enabled]
[ INFO] [1538127173.476528171]: KCL: (/rosplan_plan_dispatcher) Feedback received [1, action achieved]
[ INFO] [1538127173.577437669]: KCL: (/rosplan_plan_dispatcher) 1: action localise completed
[ INFO] [1538127173.579703540]: KCL: (/rosplan_plan_dispatcher) Dispatching action [2, goto_waypoint, 78.133855, 60.000000]
[ INFO] [1538127173.586040024]: KCL: (goto_waypoint) action received
[ INFO] [1538127173.604621464]: KCL: (/rosplan_knowledge_base) Removing domain attribute (robot_at)
[ INFO] [1538127173.607979159]: KCL: (goto_waypoint) Action completing with probability 1.000000 and duration 0.000000
[ INFO] [1538127173.610480447]: KCL: (/rosplan_knowledge_base) Adding fact (robot_at kenny wp0)
[ INFO] [1538127173.610537387]: KCL: (/rosplan_knowledge_base) Adding fact (visited wp0)
[ INFO] [1538127173.676073055]: KCL: (/rosplan_plan_dispatcher) Feedback received [2, action enabled]
[ INFO] [1538127173.676103974]: KCL: (/rosplan_plan_dispatcher) Feedback received [2, action achieved]
[ INFO] [1538127173.777394110]: KCL: (/rosplan_plan_dispatcher) 2: action goto_waypoint completed
[ INFO] [1538127173.779571656]: KCL: (/rosplan_plan_dispatcher) Dispatching action [3, goto_waypoint, 138.134862, 60.000000]
[ INFO] [1538127173.786146841]: KCL: (goto_waypoint) action received
[ INFO] [1538127173.787018334]: KCL: (/rosplan_knowledge_base) Removing domain attribute (robot_at)
[ INFO] [1538127173.787112866]: KCL: (goto_waypoint) Action completing with probability 1.000000 and duration 0.000000
[ INFO] [1538127173.788078660]: KCL: (/rosplan_knowledge_base) Adding fact (robot_at kenny wp1)
[ INFO] [1538127173.788118871]: KCL: (/rosplan_knowledge_base) Adding fact (visited wp1)
[ INFO] [1538127173.877952111]: KCL: (/rosplan_plan_dispatcher) Feedback received [3, action enabled]
[ INFO] [1538127173.878007381]: KCL: (/rosplan_plan_dispatcher) Feedback received [3, action achieved]
[ INFO] [1538127173.976284797]: KCL: (/rosplan_plan_dispatcher) 3: action goto_waypoint completed
[ INFO] [1538127173.981061031]: KCL: (/rosplan_plan_dispatcher) Dispatching action [4, goto_waypoint, 198.135854, 60.000000]
[ INFO] [1538127173.987643588]: KCL: (goto_waypoint) action received
[ INFO] [1538127173.988835307]: KCL: (/rosplan_knowledge_base) Removing domain attribute (robot_at)
[ INFO] [1538127173.989342095]: KCL: (goto_waypoint) Action completing with probability 1.000000 and duration 0.000000
[ INFO] [1538127173.990373228]: KCL: (/rosplan_knowledge_base) Adding fact (robot_at kenny wp2)
[ INFO] [1538127173.990437847]: KCL: (/rosplan_knowledge_base) Adding fact (visited wp2)
[ INFO] [1538127174.075481798]: KCL: (/rosplan_plan_dispatcher) Feedback received [4, action enabled]
[ INFO] [1538127174.075523462]: KCL: (/rosplan_plan_dispatcher) Feedback received [4, action achieved]
[ INFO] [1538127174.178690954]: KCL: (/rosplan_plan_dispatcher) 4: action goto_waypoint completed
[ INFO] [1538127174.181125108]: KCL: (/rosplan_plan_dispatcher) Dispatching action [5, goto_waypoint, 258.136861, 60.000000]
[ INFO] [1538127174.186307106]: KCL: (goto_waypoint) action received
[ INFO] [1538127174.187542487]: KCL: (/rosplan_knowledge_base) Removing domain attribute (robot_at)
[ INFO] [1538127174.187744090]: KCL: (goto_waypoint) Action completing with probability 1.000000 and duration 0.000000
[ INFO] [1538127174.188738977]: KCL: (/rosplan_knowledge_base) Adding fact (robot_at kenny wp3)
[ INFO] [1538127174.188772711]: KCL: (/rosplan_knowledge_base) Adding fact (visited wp3)
[ INFO] [1538127174.275658664]: KCL: (/rosplan_plan_dispatcher) Feedback received [5, action enabled]
[ INFO] [1538127174.275698950]: KCL: (/rosplan_plan_dispatcher) Feedback received [5, action achieved]
[ INFO] [1538127174.375693075]: KCL: (/rosplan_plan_dispatcher) 5: action goto_waypoint completed
[ INFO] [1538127174.377765821]: KCL: (/rosplan_plan_dispatcher) Dispatching action [6, goto_waypoint, 318.137868, 60.000000]
[ INFO] [1538127174.385954262]: KCL: (goto_waypoint) action received
[ INFO] [1538127174.387189693]: KCL: (/rosplan_knowledge_base) Removing domain attribute (robot_at)
[ INFO] [1538127174.387597818]: KCL: (goto_waypoint) Action completing with probability 1.000000 and duration 0.000000
[ INFO] [1538127174.388841373]: KCL: (/rosplan_knowledge_base) Adding fact (robot_at kenny wp4)
[ INFO] [1538127174.388877387]: KCL: (/rosplan_knowledge_base) Adding fact (visited wp4)
[ INFO] [1538127174.476629547]: KCL: (/rosplan_plan_dispatcher) Feedback received [6, action enabled]
[ INFO] [1538127174.476668991]: KCL: (/rosplan_plan_dispatcher) Feedback received [6, action achieved]
[ INFO] [1538127174.579087260]: KCL: (/rosplan_plan_dispatcher) 6: action goto_waypoint completed
```
and then it is stuck, so when I call `rostopic echo /rosplan_plan_dispatcher/action_dispatch -n 1` it shows me this:
```
action_id: 6
name: "goto_waypoint"
parameters: 
  - 
    key: "v"
    value: "kenny"
  - 
    key: "from"
    value: "wp3"
  - 
    key: "to"
    value: "wp4"
duration: 60.0
dispatch_time: 310.006011963
---
```
instead of empty list as shown in the tutorial.

I am using ROS kinetic.
This can be reproduce at least master as well as the kinetic branch.

gerardcanal, 2018-09-28:
Dear @brean,
I checked it and in my case, with the master branch with the tutorial 4 launch, the plan dispatch finishes:
```
(...)
[ INFO] [1538128426.933001147]: KCL: (/rosplan_knowledge_base) Adding fact (robot_at kenny wp4)
[ INFO] [1538128426.933027365]: KCL: (/rosplan_knowledge_base) Adding fact (visited wp4)
[ INFO] [1538128427.022829697]: KCL: (/rosplan_plan_dispatcher) Feedback received [6, action enabled]
[ INFO] [1538128427.022862628]: KCL: (/rosplan_plan_dispatcher) Feedback received [6, action achieved]
[ INFO] [1538128427.122816882]: KCL: (/rosplan_plan_dispatcher) 6: action goto_waypoint completed
[ INFO] [1538128427.322728539]: KCL: (/rosplan_plan_dispatcher) Dispatch complete.
```
All the actions dispatch at the same time becasue the duration of the actions is 0.

There was an issue with the plan dispatch not finishing with the esterel plan dispatcher, but this should be fixed by now in master due to #126.

As for the topic message, the result you are getting is correct. The tutorial has the wrong topic listed though. The topic showing that the action is achieved can be seen with a `rostopic echo /rosplan_plan_dispatcher/action_feedback `

@m312z maybe you can fix that in the tutorial?

m312z, 2018-09-28:
> All the actions dispatch at the same time becasue the duration of the actions is 0.

Yes -- the whole plan is meant to execute. I'll add ellipses to the output in the tutorial to indicate that it has been truncated.

> There was an issue with the plan dispatch not finishing with the esterel plan dispatcher, but this should be fixed by now in master due to #126.

Also yes -- the plan is executed, but the dispatcher hangs at the end. If you do a git pull and catkin build, then that issue should be fixed (you will see "Dispatch complete" at the end.

> The tutorial has the wrong topic listed ... "rostopic echo /rosplan_plan_dispatcher/action_feedback"

Final yes -- I'll fix that now!

Thanks both!
Michael

closed 2018-09-28 by m312z

brean, 2018-09-28:
Thank you for the quick fix! You guys are awesome!