KCL-Planning/ROSPlan #136: Tutorial 02 - generated plan go from wp0 to wp0

brean, 2018-10-10:
Hi,

We just noticed a small error in the tutorial. The generated plan tells the robot to go from wp0 to wp0 when it is actually at wp1. Shouldn't it go from wp1 to wp0? (see [Tutorial 02](http://kcl-planning.github.io/ROSPlan//tutorials/tutorial_02) )
```
0.000: (undock kenny wp1)  [10.000]
10.001: (localise kenny)  [60.000]
70.002: (goto_waypoint kenny wp0 wp0)  [60.000]
130.003: (goto_waypoint kenny wp0 wp1)  [60.000]
190.004: (goto_waypoint kenny wp1 wp2)  [60.000]
250.005: (goto_waypoint kenny wp2 wp3)  [60.000]
310.006: (goto_waypoint kenny wp3 wp4)  [60.000]
```

gerardcanal, 2018-10-10:
Hi brean,
Just to be sure, which branch, planner and domain file are you using?

brean, 2018-10-10:
I am using the master branch, which created the same file but this can also be found in the documentation so I guess it is there for quite some time.

gerardcanal, 2018-10-10:
The same happens to me. 
It's basically because of the modeling. The initial state says that the robot is at wp0, but doesn't specify that wp0 is visited. Similarly, the goto_waypoint action doesn't have any condition saying that ?from and ?to must be different. Therefore, for the planner, executing an action to go to the waypoint in which you are now is completely legal (in fact it needs to visit wp0 to acomplish the goal, so why not visiting it while you're there? :P ).

For instance, you could change the condition of the goto_waypoint to (note you also need to add the :equality requirement)
`(at start (and (not (= ?from ?to)) (robot_at ?v ?from)))` 

And then the resulting plan is:

```0.000: (undock kenny wp0)  [10.000]
10.001: (localise kenny)  [60.000]  
70.002: (goto_waypoint kenny wp0 wp1)  [60.000]  
130.003: (goto_waypoint kenny wp1 wp0)  [60.000]  
190.004: (goto_waypoint kenny wp0 wp2)  [60.000]  
250.005: (goto_waypoint kenny wp2 wp3)  [60.000]  
310.006: (goto_waypoint kenny wp3 wp4)  [60.000]
```

I hope this answers your doubt.

brean, 2018-10-10:
Thank you! I think the tutorial should be updated to include this :equality-requirement.

m312z, 2018-10-10:
Yes, agreed!

Although a nicer solution will be to either:

1) add "visited wp0" to the initial state -- it seems intuitive that the current waypoint of the vehicle is already visited.

or (2) add the precondition "(not (robot_at ?from))" which will have the same effect without requiring the equivalence (which some planners do not handle).

We could add both!

brean, 2018-10-10:
+1 for both!

oscar-lima, 2019-05-03:
issue addressed in #151 , it can be closed now

closed 2019-05-03 by oscar-lima