KCL-Planning/ROSPlan #142: Trouble with update of ROSPLAN

Samcheok, 2018-10-25:
Hello,

We notice that there is a difference between ROSPLAN version (old) downloaded in Oct. 1st  and that (new version) downloaded in  Oct 15.
Thus so,  for the same plan, Esterel plan graph generated by old version is different from that generated by new version.
The only change we are aware of is VALVisitorOperator.cpp file in  rosplan_knowledge_base/src.

Here is an sample:
[recent.pdf](https://github.com/KCL-Planning/ROSPlan/files/2513202/recent.pdf)
[old.pdf](https://github.com/KCL-Planning/ROSPlan/files/2513203/old.pdf)

Actually, old version generates correct graph for plan.

Any comments would be appreciated very much.

m312z, 2018-10-25:
Hello Kyeonseok,

Thanks for this! I think I understand the error.

Right now ROSPlan is handling all actions as durative actions. However, this has a negative effect on instant actions, which Gerard (@gerardcanal) spotted in the last pull request. See this for more information:
https://github.com/KCL-Planning/ROSPlan/pull/132#discussion_r224075445

The PDDL to Esterel plan parser will need a thorough upgrade to deal with instant actions.
1. Those actions should be encoded as a single node, with both conditions and effects attached.
2. This will necessitate a change in the dispatcher, to ensure that "action enabled" and "action achieved" feedback messages do not trigger incorrect nodes.
3. Finally, the checking for mutual exclusions should be updated to consider actions which do not have invariants between pairs of nodes.
4. At the same time, this algorithm can be made more efficient.

I'll add this issue to the project board.

Michael

jqs7328, 2018-10-26:
Dear Michael and Gerard,

First of all, thank you for your quick responses.

I'm working with Kyeongseok who posted this issue on the same project.

Our domain deals with only durative actions - 
last August we noticed that instant actions did not work on both dispatcher and esterel plan of ROSPLAN, and remodeled our domain with durative actions, only.

Thanks!

Ji-Ae Shin