KCL-Planning/ROSPlan #145: move base goal is not the same as defined in the waypoint.text file

FanhuiJ, 2018-10-26:
Hello every one,

I have a problem that I don't understand, I want to read waypoint positions in a text file, so in my lunch file, I use the rosplan_simplemap_server, bu no matter what value I put for the waypoints' position, I always get a move base goal which is [15, 20, 0], I really don't know where does it comes from. Thoses waypoints that I defined are showed correctly in Rviz, but totally wrong when sending them as move_base/goal.

I think maybe there is a problem in the rosplan_dispatch_msgs, but by showing this message, I couldn't get the position of waypoints.

In my launch file, I put this:

	<node name="rosplan_simplemap_server" pkg="rosplan_interface_mapping" type="rpsimplemapServer" respawn="false" output="screen">
	   <param name="data_path" value="$(find rosplan_planning_system)/common/" />
	   <param name="waypoint_file" value="$(find rosplan_demos)/common/waypoints_for_test.txt" />
    	</node> 

An in my waypoints_for_test.txt file, I put this:

**wp_name [0, 0, 0.234] 
wp_name2 [5, 4, 0.234]**

And what I got as move_base goal when I run the plan_and_dispatch.bash file (call servers: problem generation server, planning server, parse plan, dispatch plan):

ros@ros-vm:~/Documents/ROS$ rostopic echo /move_base/goal 
WARNING: no messages received and simulated time is active.
Is /clock being published?
header: 
  seq: 0
  stamp: 
    secs: 31
    nsecs: 710000000
  frame_id: ''
goal_id: 
  stamp: 
    secs: 31
    nsecs: 710000000
  id: "/rosplan_interface_movebase-1-31.710000000"
goal: 
  target_pose: 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "map"
    **pose: 
      **position: 
        x: 15.0
        y: 20.0**
        z: 0.0**
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.841470984808
        w: 0.540302305868
---

Can anyone give me some advices to fix the problem?
Thanks at first!

yashu88, 2018-11-06:
Hello,

Kindly check real_x  and real_y relation with x and y coordinate. 

Also check RPSimpleMapVisualization.cpp, there might be some relation between real_x , real_y coordinate with position x and y co-ordinate.

			marker.pose.position.x = wit->second->real_x;

			marker.pose.position.y = wit->second->real_y;

In case of RPRoadmapServe OccupancyGrid and Odometry handles x and y coordinate. But not sure about RPSimpleMapServer.

			geometry_msgs::Point real_point = occupancy_grid_utils::cellCenter(map_meta_data, cell);

			real_x = real_point.x;

			real_y = real_point.y;

Good luck.

oscar-lima, 2019-05-03:
rosplan_demos are now moved to: https://github.com/KCL-Planning/rosplan_demos
and there are plans to remove mongoDB from the rosplan_demos in favor of storing the waypoints in the ROS parameter server. If this issue persists please add as an issue on https://github.com/KCL-Planning/rosplan_demos

closed 2019-05-03 by oscar-lima