KCL-Planning/ROSPlan #147: Problem with popf binary

jandegier, 2018-10-30:
Hi all,

After installation of ROSPlan, I started with the ROSPlan tutorial, and I have an error in step 2 related to src/rosplan/rosplan_planning_system/common/bin/popf.

I am using the windows subsystem for linux (Ubuntu 18.04.1) and ROS melodic.

Anyway, the output is as follows:

```
$ roslaunch tutorial_02.launch
... logging to ./.ros/log/1b8d3b60-dc8b-11e8-9ead-d481d7954389/roslaunch-PC0000-7999.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://PC0000:54683/

SUMMARY
========

PARAMETERS
 * /config_manager/defaults_path:
 * /message_store/mongodb_use_localdatacenter: True
 * /message_store/queue_size: 100
 * /mongo_server/database_path: ./cat...
 * /mongodb_host: localhost
 * /mongodb_port: 62345
 * /mongodb_use_daemon: False
 * /replicator_node/replicator_dump_path: /tmp/replicator_d...
 * /rosdistro: melodic
 * /rosplan_knowledge_base/database_path: ./cat...
 * /rosplan_knowledge_base/domain_path: ./cat...
 * /rosplan_knowledge_base/problem_path: ./cat...
 * /rosplan_knowledge_base/use_unknowns: False
 * /rosplan_planner_interface/data_path: ./cat...
 * /rosplan_planner_interface/domain_path: ./cat...
 * /rosplan_planner_interface/planner_command: timeout 10 /home/...
 * /rosplan_planner_interface/planner_topic: planner_output
 * /rosplan_planner_interface/problem_path: ./cat...
 * /rosplan_planner_interface/problem_topic: /rosplan_problem_...
 * /rosplan_planner_interface/use_problem_topic: True
 * /rosplan_problem_interface/domain_path: ./cat...
 * /rosplan_problem_interface/knowledge_base: rosplan_knowledge...
 * /rosplan_problem_interface/planning_language: PDDL
 * /rosplan_problem_interface/problem_path: ./cat...
 * /rosplan_problem_interface/problem_topic: problem_instance
 * /rosversion: 1.14.3

NODES
  /
    config_manager (mongodb_store/config_manager.py)
    message_store (mongodb_store/message_store_node.py)
    mongo_server (mongodb_store/mongodb_server.py)
    replicator_node (mongodb_store/replicator_node.py)
    rosplan_knowledge_base (rosplan_knowledge_base/knowledgeBase)
    rosplan_planner_interface (rosplan_planning_system/popf_planner_interface)
    rosplan_problem_interface (rosplan_planning_system/problemInterface)

auto-starting new master
process[master]: started with pid [8009]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1b8d3b60-dc8b-11e8-9ead-d481d7954389
process[rosout-1]: started with pid [8020]
started core service [/rosout]
process[rosplan_knowledge_base-2]: started with pid [8023]
process[rosplan_problem_interface-3]: started with pid [8028]
process[mongo_server-4]: started with pid [8029]
[ INFO] [1540935081.579668400]: KCL: (/rosplan_knowledge_base) Parsing domain
[ INFO] [1540935081.586170200]: KCL: (/rosplan_knowledge_base) Parsing domain: ./catkin_ws/src/ROSPlan/src/rosplan/rosplan_demos/common/domain_turtl
ebot.pddl.
[ INFO] [1540935081.595805500]: KCL: (/rosplan_knowledge_base) Parsing initial state
[ INFO] [1540935081.598528500]: KCL: (/rosplan_knowledge_base) Parsing Problem File: ./catkin_ws/src/ROSPlan/src/rosplan/rosplan_demos/common/proble
m_turtlebot.pddl.
[ INFO] [1540935081.672929700]: KCL: (/rosplan_knowledge_base) Waiting for MongoDB
[ INFO] [1540935081.680611500]: waitForService: Service [/message_store/delete] has not been advertised, waiting...
[ INFO] [1540935081.694292600]: KCL: (/rosplan_problem_interface) Ready to receive
process[config_manager-5]: started with pid [8040]
process[message_store-6]: started with pid [8041]
process[replicator_node-7]: started with pid [8042]
process[rosplan_planner_interface-8]: started with pid [8043]
[ INFO] [1540935083.713663600]: KCL: (/rosplan_planner_interface) Ready to receive
[ INFO] [1540935090.891419500]: waitForService: Service [/message_store/delete] is now available.
MongoDB shell version v3.6.3
connecting to: mongodb://127.0.0.1:27017/message_store
MongoDB server version: 3.6.3
2018-10-30T22:31:31.413+0100 E QUERY    [thread1] Error: remove needs a query :
DBCollection.prototype._parseRemove@src/mongo/shell/collection.js:357:1
DBCollection.prototype.remove@src/mongo/shell/collection.js:382:18
@(shell eval):1:11
[ INFO] [1540935091.431961900]: KCL: (/rosplan_knowledge_base) Ready to receive
```
Everything seems fine (except for a failed remove query in mongodb). So I proceed with the tutorial, by executing
```
$ rosservice call /rosplan_problem_interface/problem_generation_server
```
resulting in:
```
[ INFO] [1540935125.747053000]: KCL: (/rosplan_problem_interface) (problem.pddl) Generating problem file.
[ INFO] [1540935125.842193100]: KCL: (/rosplan_problem_interface) (problem.pddl) The problem was generated.
[ INFO] [1540935125.845453700]: KCL: (/rosplan_planner_interface) Problem received.
```
Next step:
``` 
$ rosservice call /rosplan_planner_interface/planning_server
```
and there is no plan. Instead I get a weird error, relating to the popf binary file:
```
[ INFO] [1540935144.915754800]: KCL: (/rosplan_planner_interface) (problem.pddl) Writing problem to file.
[ INFO] [1540935144.920818400]: KCL: (/rosplan_planner_interface) (problem.pddl) Running: timeout 10 ./catkin_ws/src/ROSPlan/src/rosplan/rosplan_pla
nning_system/common/bin/popf ./catkin_ws/src/ROSPlan/src/rosplan/rosplan_demos/common/domain_turtlebot.pddl ./catkin_ws/src/ROSPlan/src/
rosplan/rosplan_demos/common/problem.pddl > ./catkin_ws/src/ROSPlan/src/rosplan/rosplan_demos/common/plan.pddl
./catkin_ws/src/ROSPlan/src/rosplan/rosplan_planning_system/common/bin/popf: 2: ./catkin_ws/src/ROSPlan/src/rosplan/rosplan_planning_sys
tem/common/bin/popf: Syntax error: ";" unexpected
[ INFO] [1540935145.128623300]: KCL: (/rosplan_planner_interface) (problem.pddl) Planning complete
[ INFO] [1540935145.130894800]: KCL: (/rosplan_planner_interface) (problem.pddl) Plan was unsolvable.
```

I have no idea how I can fix this. Does it have to do with the melodic version of ROS? Some help is much appreciated. Thanks in advance!

florianislitzer, 2019-07-04:
@jandegier Have you solved your problem?
When I try to parse i get the same error.

vijayakumar-bhaskar, 2020-04-05:
> @jandegier Have you solved your problem?
> When I try to parse i get the same error.
> 
> UPDATE:
> 
> I´m using Ubuntu Subsystem for Windows, the precompiled popf binary is not working for that system.
> After compiling popf (https://github.com/LCAS/popf) by myself it worked.

@florianislitzer Do you have the popf file, I am facing similar problem as you.