KCL-Planning/ROSPlan #148: An Error Occurred when Installing ROSPlan

Terryxyz, 2018-11-08:
Hello everyone,

I have followed the step shown in the tutorial (https://github.com/KCL-Planning/ROSPlan/blob/master/README.md) step by step to install ROSPlan. However, when I build the catkin workspace, it tells me there is an syntax error in parser.ypp as follows.

rddl_parser/parser.ypp:2.9-18: syntax error, unexpected identifier, expecting string
make[2]: *** [parser.tab.cc] Error 1
make[1]: *** [CMakeFiles/rddl_parser.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2

I have not changed anything after downloading the files from Github. Is there anything wrong with the new version file? Thanks.

Terryxyz, 2018-11-09:
Hi,

I have installed the bison package. However, the error still exists. The version of ROS is kinetic and the version of Ubuntu is 16.04. It says there is a syntax error in parser.ypp.

gerardcanal, 2018-11-09:
Thanks for the answer.
Can you show me 3 or 4 lines that appear before the error? Also, can you confirm me which bison version is installed in your system? 
I can't reproduce the error.

gerardcanal, 2018-11-09:
Also, as I cannot reproduce, can you try to change the line 2 of the file parser.ypp from
`%define api.prefix {rddl_yy}` to `%define api.prefix "rddl_yy"` and try to recompile?

gerardcanal, 2018-11-11:
Ok, it's a strange one... I tried in many machines and was not able to reproduce.
So in order for you to be able to to go on with ROSPlan and not be blocked on this I'd suggest you use the "kinetic" branch, which is stable.

However, I'd appreciate if you could provide us more information to fix this issue (as master will end up merged to the kinetic branch at some point).

Could you attach the output of the `catkin build -v` as I can't see the output of the whole cmake without the -v option. Please ensure to clean before by doing a `catkin clean` before. 
Also, please try to use the catkin_make command to see if that changes something (I doubt it though...)

Thank you for the help! :)

Terryxyz, 2018-11-11:
Hi!

I have found where the problem is. I change the command to  git clone -b kinetic https://github.com/KCL-Planning/ROSPlan.git , the compilation succeeds. Thank you.

gerardcanal, 2018-11-11:
Hi Terryxyz,
Indeed it compiles because it doesn't have the changes that fail to compile. (That's what I meant when I told u to use the kinetic branch). But the master branch will be merged to the kinetic one soon so the problem will be back there.

Could you try to go to master branch and do the things I asked? It would help me to find the problem :)

Thank you!

Terryxyz, 2018-11-12:
Hi,

I have tried to go to master branch, the problem exists as before. I have checked that the version of bison is just 2.3 and I cannot upgrade it to a newer version. The result of  catkin build -v is so long and it is difficult to show here. Maybe you can tell me how to get bison of the version 3.0.4 and I will try again. Thanks.

gerardcanal, 2018-11-12:
Okay, that's what I suspected. 
To me, it looks like your system has two versions of bison, the one from the Ubuntu repositories (which you told me was 3.0.4) and another one which is 2.3.
You can try to change the line 56 of the CMakeListst.txt file from the rosplan_dependencies package from 
`find_package(BISON REQUIRED)` to ` find_package(BISON 3.0.4 REQUIRED)`, see if with this the CMake can find the correct one. If this doesn't work, you should try to find where is the 2.3 version of bison installed and uninstall it (or install 3.0.4 from sources).

I hope this helps! In the meantime, you can work with the kinetic branch which will work for you! :) 
Thank you.

closed 2018-11-12 by Terryxyz